lowRISC/ibex #1: migrating to micro-riscy configuration problem

abdallah93as, 2017-08-21:
Hello I am having a bit of trouble configuring the processor to work as "micro-riscy"

I changed all the: 
parameter RV32M = 0  instead of 1
and
parameter RV32E = 1 instead of 0

But I still see mult_div signals in the modelsim waveform.

Is there anything else that I need to do in order to configure to that version.

Thankyou

FrancescoConti, 2017-08-21:
Do you see any activity on them? I don't think it is a problem that the signals are there at all, however if all parameters are set correctly they will be unused.

abdallah93as, 2017-08-21:
IT WORKED!!!

I assumed that if I set these parameters correctly, than I would no longer see there respective signals.

I did a vivado implementation to see if I'm really in micro-riscy mode, and when the implementation was done, I saw that the number of flip-flops have gone down more than half of what they were in zero-riscy.

GREAT JOB GUYS, I really like the two new implantations. Keep up the good work :D

closed 2017-08-21 by abdallah93as