lowRISC/ibex #2: Support for riscv-formal

chaosbastler, 2018-01-02:
Is it possible to support the [RISC-V Formal Verification Framework](https://github.com/cliffordwolf/riscv-formal)? If yes, are there plans to do this?

davideschiavone, 2018-01-03:
not yet, but we would be very glad if someone wants to implement it

wallento, 2018-10-12:
Hi, we have started working on this. We hope to have it ready by end of October.

wallento, 2018-10-12:
Feel free to assign to me

raulbehl, 2019-05-02:
Hi @wallento, 

I know it's been a long time but just wanted to know if you were able to get the formal framework done?

imphil, 2019-05-16:
@towoe that's something you worked on, right? Do you plan to get the necessary changes included upstream (at least the interface, we need to think a bit more about the Verilog parser workarounds)?

towoe, 2019-05-16:
Yes I worked on this and plan to upstream it. I guess I will need to do some work as the code changed a bit.

raulbehl, 2019-05-16:
@towoe Is it available on a branch which I could use?

towoe, 2019-05-17:
@raulbehl not yet sorry. But will try to push it soon!

towoe, 2019-06-05:
I pushed to https://github.com/towoe/ibex/tree/rvfi
But this will not yet work with the framework if yosys is used, as ibex makes use of systemverilog features which are not supported by yosys. I have a fork of yosys where I incorporate some work which is already available as a PR, but this not enough to get all issues resolved. One can manually modify ibex to get it working though. This will include removing structs, specifying enums which are imported, type changes, loop changes, removal of 2D arrays and more. This can all be done without changing the behavior, but will decrease the code quality.

imphil, 2019-06-05:
My preferred way forward would be:
1. Merge the RVFI 
2. Switch ibex_tracer.sv to use it
3. PR and merge the riscv-formal things as soon as we know it at least works with the Verific frontend. (And I can't test this.)

@towoe could you do a PR based on your changes for the RVFI (only), and if you have time switch the ibex_tracer.sv to use it?

towoe, 2019-06-05:
@imphil I created the PR https://github.com/lowRISC/ibex/pull/61
Is this what you meant by including "(only)" RVFI?

imphil, 2019-06-05:
Yes, thanks!

towoe, 2019-06-18:
I moved the `riscv-formal` specifics into https://github.com/towoe/riscv-formal/tree/ibex

ivanmgribeiro, 2019-07-10:
Hi, I was wondering if there was still ongoing work on implementing the RVFI signals? It seems some of the signals are not behaving as they should - the rvfi_halt signal for example is hardwired to 0.

imphil, 2019-07-10:
@ivanmgribeiro all outstanding patches have been merged. If there are issues with the current code, please open a new issue or (even better) a pull request. We currently cannot test if the signals are actually working in RISC-V Formal, as we don't have a license for the Verific backend needed to compile the code. But we're of course happy to fix things!

towoe, 2019-07-11:
Hi @ivanmgribeiro yes the `rvfi_halt` and also `rvfi_trap` are not connected yet. I have not implemented this in the first iteration, and as @imphil said we do not have the Verific backend, so developing and testing it further is not possible. If there is a signal which does show an obvious misbehavior we can check this though.

asb, 2019-07-11:
Although ability to run the riscv-formal flow is a good test, we will ideally have more targeted unit tests for the interface independent of that. I know @imphil is interested in setting up some Cocotb-based verification, so this might be a good area to try it out on.

towoe, 2019-07-17:
I updated RVFI in PR https://github.com/lowRISC/ibex/pull/155 as a result of changes to the core. As we are still not able to test it, I would assume that there are still differences to the riscv-formal requirements, but the tracer uses the interface now https://github.com/lowRISC/ibex/pull/77.

towoe, 2019-12-12:
The interface needs to be updated to support [CSRs](https://github.com/SymbioticEDA/riscv-formal/blob/master/docs/rvfi.md#control-and-status-registers-csrs).

imphil, 2020-10-23:
@towoe I know you have been playing around with RISC-V Formal before and from what I remember it should work? The interface is certainly stable and well-tested these days, anything else missing?

towoe, 2020-10-30:
From the Ibex side the support is basically there.
I wanted to see if it is possible to improve the build flow (sv2v/Yosys) to reuse the FuseSoC approach, but did not succeed the last time I tried. The easiest option would be to have the yosys-without-sv2v flow you mentioned in #1006 
And the support is not yet in riscv-formal, still need to make the PR, but wanted to improve on build flow before.
So maybe we keep this issue open until Ibex is fully integrated with riscv-formal one way or the other.

imphil, 2020-10-30:
There seems to be no interest or time from the yosys maintainers to review and merge the remaining patches to get Ibex compiling in Yosys (see https://github.com/lowRISC/ibex/issues/60). This means we'll need to rely on a patched version of Yosys for the timing being. I'm planning to add it to CI soon-ish, and it would make sense to base other work on top of it as well. So if you can get riscv-formal working with Yosys + the patches mentioned in #60 I'd appreciate a PR that finishes the flow around that version of Yosys.

towoe, 2020-11-03:
I would prefer to use a patched Yosys version. Removing the sv2v conversion reduces the complexity when interacting with riscv-formal. I will do a WIP PR which depends then on a correct patched Yosys version which I did not yet find as mentioned in #60.