lowRISC/ibex #3: la register not update when single step debug

timdudu, 2018-02-09:
When core execute JAL instruction, return address should be written to la register. But in single step execution, the la register is not updated.
Here is an example waveform. 
code is:  730000ef                jal     ra,102073c

1. the core is free run.
[image]
You can see the return address 0x1020010 is written to la register.

2. the core is in single step.
[image]
Return address 0x1020010 is not written to la register.  We debug this issue and found that core deassert we_a_i in DBG_SIGNAL state. And ctrl_fsm_ns enter DBG_SIGNAL state before la register update. 
I tried to fix this issue by add & id_ready_i when ctrl_fsm_ns try to enter DBG_SIGNAL.
              jump_set_i: begin
                pc_mux_o          = PC_JUMP;
                pc_set_o          = 1'b1;
                perf_jump_o       = 1'b1;
                dbg_trap_o        = dbg_settings_i[DBG_SETS_SSTE];
                if (dbg_req_i & id_ready_i)
                  ctrl_fsm_ns = DBG_SIGNAL;

The return address can be written to la due to this change. But I'm not sure this is a good workaround.
[image]

RI5CY core also has the same issue.

timdudu, 2018-02-09:
Sorry, it's ra register not la.

davideschiavone, 2018-02-09:
Thanks a lot Tim,

I'll fix it asap and let you know soon.

Thanks a lot for reporting it ,
Davide

davideschiavone, 2018-02-16:
Dear Tim, 

I fixed the bug, you can find the fix here: d4d46971ebd5e0a637d2d182af675e7c43078885

The solution is similar to the one you proposed but I simply removed the check of dbg request as it will be check in the second cycle of the instruction.

Thanks a lot,
Davide

closed 2018-02-16 by davideschiavone