lowRISC/ibex #8: Support for PMP

wallento, 2018-10-12:
Implement Physical Memory Protection (PMP) as per privspec 1.10. My team is planning to work on that.

davideschiavone, 2018-11-09:
It's currently under development for Riscy in the branch MPU_branch.
once it is stable and well tested it will be merged into the master.
Then we can try to port it to zero-riscy.
Any help is welcome of course.

Best
Davide

imphil, 2019-08-06:
@towoe did you already start implementing PMP in the past?

towoe, 2019-08-06:
No sorry, haven't worked on that.

wallento, 2019-08-06:
I did work on it and finally have access to my old work computer. Will check about it tomorrow!

vogelpi, 2019-08-30:
PMP support has been added by @tomroberts-lowrisc as part of lowrisc/ibex#252. I am thus closing the issue.

closed 2019-08-30 by vogelpi