lowRISC/ibex #13: Issues using Zero-Riscy in Vivado

supersnackbros, 2019-01-09:
I apologize in advance if this is the wrong place to post this, but I could not think of anywhere else. I intend to use the Zero-Riscy core as part of a Xilinx Vivado project, but I have encountered two issues that I was hoping someone might be able to answer.

1. When running behavioral simulation and synthesis, several errors occur: "illegal context for assignment pattern" for lines 172 - 174, 181, 182, 189, 190, 197, 204, 211, 212, 222, 225, 256, 260, 284, and 285 in 'zeroriscy_tracer.sv'. I am more familiar with Verilog than SystemVerilog and searching for solutions provided no answers. What should I do?

2. After reading the documentation, thoroughly, I am still unsure how to start the core's main operation. I've set 'clk_i' to a steady clock and set both 'clock_en_i' and 'fetch_enable_i' positive as part of my testbench. What is it that I have missed?

Thank you so much and sorry for your trouble.

brabect1, 2019-01-19:
Hi there,

As for 1. I believe the tracer code is purely for simulation, but is not really needed even for that. For synthesis you shall define the SYNTHESIS macro (either in one of the include files or in Vivado synthesis and simulation settings); then it should synthesize and simulate fine.

If you wanted some Vivado kick start, you may check my Digilent Arty A7-35 demo https://github.com/brabect1/zr-soc/tree/master/fpga/arty-a7 (look in `scripts`). It might also help you with the 2nd question of yours (look in https://github.com/brabect1/zr-soc/tree/master/rtl).

supersnackbros, 2019-01-20:
Thank you for the advice on the first part. Synthesis works fine now. However, I'm still having issues getting the CPU to work in the Vivado simulation. What I've done is packaged the ZR core as an IP and placed it into a block diagram. I created block diagram ports for each input/output and created an HDL wrapper for the design. I am trying to use a testbench file to perform behavioral and post-synthesis functional simulation so I can know without a shadow of a doubt how it works. My problem is that after setting the clock to a regular cycle and the clock enable and fetch enable signals to positive, the core does not seem to be reacting to the instruction grant or instruction read valid signals that I feed it. Am I going about this completely wrong?

oholimoli, 2019-05-07:
Hello, is it working now? Do you have your Zero-Riscy/Ibex Vivado-Project on GitHub?

imphil, 2019-05-07:
You can use pulpissimo with zero-riscy, which should be synthesizable using Vivado (https://github.com/pulp-platform/pulpissimo). In general, the code in this repository works with Vivado, but it's of course only a core and not a full SoC -- you need to either do this yourself, or use an existing one like pulpissimo.

closed 2019-05-16 by imphil