lowRISC/ibex #14: ASIC synthesis

Liyang131313, 2019-01-10:
Hello，How many MHz can the zero-riscy ASIC synthesis achieve? @zarubaf @Razer6 @wallento @fabianschuiki

daviderossi1982, 2019-01-11:
Up to 450 MHz in TSMC CMOS 40nm LP technology when integrated in a PULPissimo system (the SoC part of Mr. Wolf as described in this paper: https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=8494247) Actually most of the critical path is on the memory in this chip, hence you can get even higher frequency if you couple the core with simple data and instruction memories. Maybe @davideschiavone has also updated information about the synthesis of the core itself.

davideschiavone, 2019-01-11:
It can reach more than 560MHz in UMCL65 technology.

If you need more details about the physical implementation I suggest you to read the following paper: 
https://ieeexplore.ieee.org/stamp/stamp.jsp?arnumber=8106976

If you have further questions do not hesitate to ask them!

Best
Davide

closed 2019-05-07 by imphil