lowRISC/ibex #18: dbg_req_i & irq_req_ctrl_i both are high

darbaria, 2019-03-07:
Following assertion in zeroriscy_controller (Line 515) is failing in 12 cycles after reset in formal. I've turned on the PULP_FPGA_EMUL to not use clock gating for now to check this. Can send the VCD but not allowed to attach here. 

assert property (
    @(posedge clk) (~(dbg_req_i & irq_req_ctrl_i)) ) else $warning("Both dbg_req_i and irq_req_ctrl_i are active");

darbaria, 2019-03-07:
stall_cs is in HALT_REQ mode causing dbg_o to become high (line 375 in zeroriscy_debug_unit module), while at the same time, exec_ctrl_cs is in RQ_PENDING state, causing irq_req_ctrl_o (line 52 in zeroriscy_int_controller module ) to go high in the same clock cycle. Looks like an RTL bug to me!

davideschiavone, 2019-03-08:
Thanks for underlining this situation.

I don't think it is a bug but rather an deprecated assertion statement. If both are high, the interrupts have priorities. In single step the debugger would see the first instruction of the ISR as next instruction.

best regards
Davide

darbaria, 2019-03-08:
Hi Davide
Thank you for getting back, so I’ll disable this assertion for now and will look at checking priority for this. I do have other assertion failures in the LSU (other than the last one all three fail towards the very end of the lsu module). Could these be obsolete assertions as well? As I said I’m happy to share screen and show the waves. 
Thanks Ashish
    	

    	Get Outlook for iOS

davideschiavone, 2019-03-08:
Hello Ashish,
Thank a lot, yes they may be bugs even though they are more or less or related to inputs signals no?

Thanks for sharing :) very kind of you

Best
Davide

darbaria, 2019-03-11:
Hi Davide

So yes not sure why the following is an assertion, as it does need to be a constraint as data_rvalid_i is a top-level input.
 assert property (
    @(posedge clk) (CS == IDLE) |-> (data_rvalid_i == 1'b0) );
However, the following assertion doesn't seem to reflect what the intent (in the comment) is. The comment says that it should not be possible to get a grant without an rvalid for the last request but the assertion doesn't reflect that. In any case this assertion fails, and I believe this should also be an assumption as data_gnt_i is an input. I'll fix this but I've a question:

Is it true that without a pending data_rvalid_i we cannot have a data_gnt_i; and that these are precisely paired; as in once a new data_rvalid_i has been serviced by a corresponding data_gnt_i then you cannot have a new data_gnt_i without a new data_rvalid_i? 

A

davideschiavone, 2019-03-11:
Hello Ashish,

I see your point indeed. In practice once you send a request, you can receive the grant from that cycle on, no matters how many cycles later. But the valid signal must come at least 1 cycle after the grant. So if you receive the grant, you can send the next request but the next grant must come together or after the valid related to the previous transaction.

I think this was the intent of the assertion.

Does it make sense for you?
Davide

vogelpi, 2019-08-13:
These signals/interfaces and the controller have been reworked. The assertion has been removed. Ibex is now using an external debug module following the RISC-V Debug Spec version 0.13. I am thus closing this issue.

closed 2019-08-13 by vogelpi