lowRISC/ibex #21: Testbench for running simulations

raulbehl, 2019-05-02:
Hi, 

I was going through the source directory but couldn't find an existing Testbench which could be used to run sims. Is there a chance that the TB would also made open-source?

FrancescoConti, 2019-05-02:
Hi, we don’t have exactly an open source standalone testbench for Zero-Riscy, but you can
- use PULPissimo as a simulation platform
- use PULPino as a simulation platform (but beware, the SW environment is legacy)
- slightly modify other testbenches. Here (https://github.com/pulp-platform/hwpe-tb) you find a testbench for accelerators which also includes a Zero-Riscy core and dummy memories

raulbehl, 2019-05-02:
Thank you for the comment. I'll take a look at them. 

Closing!

closed 2019-05-02 by raulbehl