lowRISC/ibex #24: Fix PDF documentation build

imphil, 2019-05-03:
```cmd
$ LANG=C make latexpdf
Running Sphinx v2.0.1
building [mo]: targets for 0 po files that are out of date
building [latex]: all documents
updating environment: 12 added, 0 changed, 0 removed
reading sources... [100%] register_file                                                                                                                                                      
looking for now-outdated files... none found
pickling environment... done
checking consistency... done
processing ibex.tex... index introduction integration instruction_fetch load_store_unit register_file cs_registers interrupts performance_counters exception_interrupts debug copyright 
resolving references...
done
writing... failed

Exception occurred:
  File "/home/philipp/src/ibex/doc/venv/lib64/python3.7/site-packages/cairosvg/parser.py", line 414, in __init__
    'No tag with id="{}" found.'.format(element_id))
TypeError: No tag with id="vvv-2-2" found.
The full traceback has been saved in /tmp/sphinx-err-u9zes32v.log, if you want to report the issue to the developers.
Please also report this if it was a user error, so that a better error message can be provided next time.
A bug report can be filed in the tracker at <https://github.com/sphinx-doc/sphinx/issues>. Thanks!
make: *** [Makefile:20: latexpdf] Error 1
```

The SVG image in question actually is generated through wavedrom, as this part of the stacktrace shows:

```
  File "/home/philipp/src/ibex/doc/venv/lib64/python3.7/site-packages/sphinxcontrib/wavedrom.py", line 187, in visit_wavedrom
    imgname = render_wavedrom(self, node, outpath, bname, format)
  File "/home/philipp/src/ibex/doc/venv/lib64/python3.7/site-packages/sphinxcontrib/wavedrom.py", line 162, in render_wavedrom
    cairosvg.svg2pdf(svgout.tostring(), write_to=fpath)
```

The SVG (which is then converted to PDF and causes the error) contains a couple of tags with `xlink:href="#vvv-2-2"` statements, but no tag with the `id="vvv-2-2"` attribute. So the error message of cairosvg is correct, and something is wrong in the generation of the wavedrom SVG (for reference: the same problem happens with `vvv-2`, but `vvv-3` is handled correctly in the SVG)

The following wavedrom source code is trying to be rendered. It's in [doc/instruction_fetch.rst](https://github.com/lowRISC/ibex/blob/master/doc/instruction_fetch.rst):

```
{"signal":
  [
    {"name": "clk", "wave": "p......"},
    {"name": "data_addr_o", "wave": "x===xxx", "data": ["Address", "Address", "Address"]},
    {"name": "data_req_o", "wave": "01..0.."},
    {"name": "data_gnt_i", "wave": "0..10.."},
    {"name": "data_rvalid_i", "wave": "0....10"},
    {"name": "data_rdata_i", "wave": "xxxxx=x", "data": ["RData"]}
  ],
  "config": { "hscale": 2 }
}
```

So we have:

1. A bug: wavedrom should not generate invalid SVG.
2. More to investigate: is the wavedrom source actually wrong, or is the conversion wrong?
3. More error reporting would be helpful, getting the file which causes the error required adding print statements across the sources.

wallento, 2019-05-13:
Hi,

the problem is related to a recent partial rewrite of the pre-rendering. Seems there is an issue with value-to-value transitions. Will get to it soon.

Cheers

wallento, 2019-05-14:
https://github.com/sphinx-doc/sphinx/pull/6364

wallento, 2019-05-14:
#36 and #37 make the documentation build. The LaTeX still looks ugly as the tables need to be set properly.

PR #37 (2019-05-16): Temporarily switch to custom sphinx
Using figures in admonitions did not work so far. A patch was created
and merge is pending. As long as it is not merged, use custom sphinx.

Fixes #24
commit 62185d05 (2019-05-16): Temporarily switch to custom sphinx

Using figures in admonitions did not work so far. A patch was created
and merge is pending. As long as it is not merged, use custom sphinx.

Fixes #24
closed 2019-05-16 by imphil