lowRISC/ibex #30: Cleanup: Remove REG_* defines from ibex_id_stage

imphil, 2019-05-08:
Can we get rid of those defines and replace them by parameters or structs?

wallento, 2019-05-17:
I was wondering about the same topic when I looked at the muldiv_fast yesterday.

`ibex_defines` is supposedly the best way to store them.

All occurrences of defines in the RTL code:

```
ibex_cs_registers.sv:  `define MSTATUS_UIE_BITS        0
ibex_cs_registers.sv:  `define MSTATUS_SIE_BITS        1
ibex_cs_registers.sv:  `define MSTATUS_MIE_BITS        3
ibex_cs_registers.sv:  `define MSTATUS_UPIE_BITS       4
ibex_cs_registers.sv:  `define MSTATUS_SPIE_BITS       5
ibex_cs_registers.sv:  `define MSTATUS_MPIE_BITS       7
ibex_cs_registers.sv:  `define MSTATUS_SPP_BITS        8
ibex_cs_registers.sv:  `define MSTATUS_MPP_BITS    12:11
ibex_id_stage.sv:`define REG_S1 19:15
ibex_id_stage.sv:`define REG_S2 24:20
ibex_id_stage.sv:`define REG_D  11:07
ibex_multdiv_fast.sv:`define OP_L 15:0
ibex_multdiv_fast.sv:`define OP_H 31:16
ibex_tracer.sv:`define REG_S1 19:15
ibex_tracer.sv:`define REG_S2 24:20
ibex_tracer.sv:`define REG_S3 29:25
ibex_tracer.sv:`define REG_D  11:07
```

imphil, 2019-05-17:
Good catch @wallento. Yes, those MSTATUS_* defines should go as well and be replaced by an mstatus enum most likely, making these bit defines obsolete.

imphil, 2019-07-29:
The MSTATUS_* defines are gone in 5b63ae9.

raulbehl, 2019-07-29:
How about those REG_* defines? Would it be worth adding an `ibex_defines.svh` file for such instruction index defines?

bp-ouhaha, 2020-06-17:
If I'm correct, I only find REG_* in the ibex_decoder.sv

We could just define this (permanent assign):
  // Source/Destination register instruction index
  logic [4:0] instr_rs1;
  assign instr_rs1 = instr[19:15];
  logic [4:0] instr_rs2;
  assign instr_rs2 = instr[24:20];
  logic [4:0] instr_rs3;
  assign instr_rs3 = instr[31:27];
  logic [4:0] instr_rd;
  assign instr_rd = instr[11:07];

and them replace all occurence below referring to the old REG_S[123] and REG_D

That would get us rid of the defines.

imphil, 2020-06-17:
> If that's ok, I'll make the Pull Request.

Sure, I'm looking forward to it!

bp-ouhaha, 2020-06-17:
maybe better to do it in 1 line with "wire" instead of "logic" ?

commit 4eece988 (2020-06-18): [rtl] Remove use of `define in decoder

Fixes #30
closed 2020-06-18 by GregAC