lowRISC/ibex #33: Enum assignment cast

towoe, 2019-05-10:
In the following lines the enum types are assigned with a value without an explicit cast
https://github.com/lowRISC/ibex/blob/93b0b77f2717a69f87608929f7cde9d5268ca2e8/rtl/ibex_controller.sv#L159

https://github.com/lowRISC/ibex/blob/93b0b77f2717a69f87608929f7cde9d5268ca2e8/rtl/ibex_controller.sv#L162

vogelpi, 2019-05-10:
Hi @towoe ,
thanks for reporting the issue. I am currently fixing and testing it.

vogelpi, 2019-05-10:
Allright, I fixed it and Vivado does not complain anymore. The patch ended up in pull request #32 . To get it working locally before the pull request is merged, you can just replace the RHS of the two assignments by `exc_cause_e'('0);` See here: https://github.com/lowRISC/ibex/pull/32/commits/6554d6726b3ab2561a5e4692af40c4a421225d75

towoe, 2019-05-10:
Thanks @vogelpi for the fix!

ikarageo, 2019-05-11:
Hi. The same issue is true for the following line:  https://github.com/lowRISC/ibex/blob/93b0b77f2717a69f87608929f7cde9d5268ca2e8/rtl/ibex_cs_registers.sv#L376

vogelpi, 2019-05-13:
Hi @ikarageo ,
thanks for reporting this issue. The corresponding statement should be:

      dcsr_q     <= '{
        xdebugver: x_debug_ver_e'('0),
        cause:     exc_cause_e'('0),
        prv:       PRIV_LVL_M,
        default:   '0
      };

vogelpi, 2019-05-13:
I created a seperate PR #35 to merge the fixes into upstream soon.

closed 2019-05-14 by vogelpi