lowRISC/ibex #46: Decode HINT as NOP

imphil, 2019-05-21:
The core fetched an compressed instruction like below.

From a downstream bug report:

core   0: 0xffffffff8000f2ec (0x00009056) add     zero, zero, s5

0x9056 is correctly decoded as HINT instruction by spike, but ibex decode it as illegal C.EBREAK instruction:

https://github.com/lowRISC/stwg-base/blob/master/hw/vendor/lowrisc_ibex/rtl/ibex_compressed_decoder.sv#L215

```
              if (instr_i[11:7] == 5'b0) begin
                // c.ebreak -> ebreak
                instr_o = {32'h00_10_00_73};
                if (instr_i[6:2] != 5'b0) illegal_instr_o = 1'b1;
```

This code first checks if instr[11:7] is zero to determine if it's c.ebreak instruction, then check instr[6:2] to see if it's illegal. However, when instr[6:2] == 0 and instr_i[11:7] != 0, it should be treated as HINT instruction rather than illegal instruction.

MACVTi, 2019-05-22:
If rs1/rd is equal to zero, then the processor should call an EBREAK instruction. I think it is spike that is decoding incorrectly here as it is bits [11:7] that take priority over bits [6:2]. I think the decoder is working correctly here.

[image]

imphil, 2019-05-22:
I had a closer look at the spec and I'm pretty confident that bits [6:2] must be handled with priority. I've opened PR #48 with an implementation.

MACVTi, 2019-05-22:
[image]

[image]

Yeah, looking at it again you're right. Your PR seems like it would correct the issue properly.

PR #48 (2019-05-23): Compressed hints
Fix handling of HINTs in compressed instructions. 

Fixes #46 (and a bit more)

@taoliug can you re-test with these modifications?
@asb does my reading of the spec sound about right?
closed 2019-05-23 by imphil