lowRISC/ibex #60: [tracking] Synthesize Ibex with Yosys

imphil, 2019-06-05:
*This is a tracking bug. This description will be updated to reflect the current state of affairs. Comment if you have suggestions or updates.*

[Yosys](http://www.clifford.at/yosys/) currently supports only a limited subset of SystemVerilog, which isn't sufficient to parse Ibex. This issue tracks the outstanding issues. We are generally open to find a common ground between tools, but only as long as it doesn't affect code quality and correctness. This means some issues will be resolved upstream in Yosys, while others might be worked around in Ibex.

## Packed multidimensional arrays not supported

### Example
```verilog
logic [7:0][31:0] a;
```

### Code workaround
Single-dimensional arrays
```verilog
logic [8*32-1:0] a;
```
Possibly memories?
```verilog
logic [31:0] a [8];
```

### Yosys bugs
- YosysHQ/yosys#340
- YosysHQ/yosys#667

## Enum  / Typedef not supported 

### Example
```verilog
typedef enum logic[1:0] { D = 2'b11, C = 2'b10, B = 2'b01, A = 2'b00 } ALP;
```

### Code workaround
params; defines

### Yosys bugs
- YosysHQ/yosys#248
- YosysHQ/yosys#298

## Assert property sensitivity list not supported 

### Example
```verilog
assert property ( @(posedge clk) ...)
```

### Workaround
- Do not use assert (for formal verification) (We disable asserts already for Verilator, we can just make the define a bit more generic for all tools which don't understand these assertions.)
- (not tested) use immediate assertions

## Int types not supported 

### Example
```verilog
int unsigned i;
```

### Code workaround
```verilog
integer i;
```

## Loop variable definition not at statement allowed

### Example
```verilog
for (integer i=0; i<5; i++) ...
```

### Code workaround
```verilog
integer i; ... for (i=0;...
```

## Break keyword not supported 

```verilog
for (...) if (a==b) break; ... 
```

### Code workaround
use variable which if set to not enter loop part again

## Package parameter keyword not allowed

```verilog
package p; parameter A = 2'b00;  endpackage 
```

### Code workaround
Use local param

Thanks @towoe for compiling this list!

imphil, 2019-06-05:
Looking at this list, we can most likely work around all Yosys problems **except for** the package parameters and the enum/typedefs. These need to be fixed upstream.

The package parameter problem doesn't seem to have an upstream bug report yet. @towoe since you've worked on this, would you mind filing one upstream?

towoe, 2019-06-05:
@imphil regarding the package parameter I just did a very simple addition in https://github.com/towoe/yosys-sv/commit/88b1e84167f44ea6e375d162026ba06a424af319
I am however not sure if this is correct or breaks anything else, but it seemed to work at my end.
As this is so small, maybe a PR to yosys can quickly answer that?

imphil, 2019-06-05:
Let's give it a try!

towoe, 2019-06-05:
As I wanted to split out the commit I just realized that this has dependencies to the other commits, I have to check what effects this has.

towoe, 2019-10-08:
Link to #342

imphil, 2020-08-20:
@daveshah1 @clairexen for visibility.

Yosys and Ibex are extremely close to the finish line, thanks for all your work so far! Let us know if there's something we can help with. We'll certainly ensure we have full CI on Ibex with Yosys to ensure nothing breaks again once all remaining patches are merged into Yosys.

towoe, 2020-11-03:
The PRs mentioned in this issued are not sufficient as far as I can see to support Ibex without any changes ([the commits applied in one branch](https://github.com/towoe/yosys-sv/commits/ibex-patches)).
If we do not want to change our code support for
 - Int types for parameters and localparam
 - Packed multi-dim arrays

are still missing.
For [ibex-yosys-build](https://github.com/antmicro/ibex-yosys-build) the branch [ibex-wip](https://github.com/antmicro/yosys/tree/ibex-wip) is used and without an exhaustive comparison there is some more work which is not yet part of any PR.
But this branch is also not producing a correct result (mult-dim conversion incorrect, struct initialization not supported).
There is some work activity, maybe some contain already the enhancements required.