lowRISC/ibex #68: Missing prim_clock_gating module

darbaria, 2019-06-06:
I can't seem to find this module anywhere. Can someone point me to where this might be?

tjaychen, 2019-06-06:
You can find it in hw/ip/prim/abstract.  It then picks either the RTL or
xilinx version underneath.

darbaria, 2019-06-06:
where exactly is hw/ip/prim/ folder?

ikarageo, 2019-06-06:
You can create it yourself as:

```
module prim_clock_gating
  (
    input  logic clk_i,
    input  logic en_i,
    input  logic test_en_i,
    output logic clk_o
  );

`ifdef FPGA_SYNTHESIS
  assign clk_o = clk_i;
`else
  logic clk_en;

  always_latch
  begin
     if (clk_i == 1'b0)
       clk_en <= en_i | test_en_i;
  end

  assign clk_o = clk_i & clk_en;
`endif

endmodule
```

darbaria, 2019-06-06:
Thanks, @ikarageo.  Oh! I didn't realize that. I do have a legacy cluster_clock_gating module from 0riscy.  BTW is it worth leaving a template .v inside the rtl folder so at least that checkout compiles with at least formal tools. I recall from 0riscy checkout I'd the same problem back then. 

@tjaychen  I'd still be interested to know where is the hw/ip folder?

tjaychen, 2019-06-06:
Oops sorry, I'm pretty new to this, it seems like I may have been talking
about the wrong thing.
Sorry about that.

towoe, 2019-06-06:
But there should be one in `dv/uvm/tb/prim_clock_gating.sv` right?

darbaria, 2019-06-06:
There is one there indeed! Should it be there?

taoliug, 2019-06-07:
I added "dv/uvm/tb/prim_clock_gating.sv" just to get the TB compile and running.  I agree we should have some default implementation like this so that user can check out the repo and get it up and running without adding an additional module.

imphil, 2019-06-07:
prim_clock_gating should be supplied by the user from his/her technology library, as documented here: https://ibex-core.readthedocs.io/en/latest/register_file.html#latch-based-register-file 

I'll keep this bug around to explore what we can do to support a sample/default implementation.

imphil, 2019-06-26:
This requirement is now more clearly documented in https://ibex-core.readthedocs.io/en/latest/getting_started.html.

closed 2019-06-26 by imphil