lowRISC/ibex #70: Exceptions handling

davideschiavone, 2019-06-07:
Hi guys (@vogelpi , @imphil )

according to the Priv ISA Spec of RISC-V 1.10, you should handle the exceptions by always jumping to mtvec rather than vectoring them as it was in the past

Look at the section
3.1.12 Machine Trap-Vector Base-Address Register (mtvec)
of this document please: https://riscv.org/specifications/privileged-isa/

I have already done it in RISCY.
So now in the controller everytime you get an exception you set exc_pc_mux_o to exceptions:
https://github.com/pulp-platform/riscv/blob/master/rtl/riscv_controller.sv#L731

and in the IF stage you always jump to the same address regardless the exception cause:
https://github.com/pulp-platform/riscv/blob/master/rtl/riscv_if_stage.sv#L141

asb, 2019-06-07:
Thanks for the note Davide - I believe @vogelpi is already working on this. We should have set up a tracking issue in this repo for that work. This issue can now fulfill that role!

taoliug, 2019-06-07:
Does this apply to interrupt handling as well?

vogelpi, 2019-06-07:
Thanks @davideschiavone , I am working on this as mentioned by @asb . Thanks for the pointers to RI5CY. This will be helpful.

@taoliug : For interrupts we have two options. Either we do it as for the exceptions (direct mode). Or we use vectored mode (similar to what we do today). The plan is to discuss this also next week in Zurich.

davideschiavone, 2019-06-07:
For what concerns interrupts, RI5CY is vectorized by default, so there is one field in the mtvec hardwired at VECTORIZED interrupt FYI.

taoliug, 2019-06-07:
The base address of the vectorized interrupt should come from MTVEC, right now it's hardwired to 0? Here's what's said in the spec:

> When MODE=Vectored,all synchronous exceptions into machine mode cause the pc to be set to the address in the BASE field, whereas interrupts cause the pc to be set to the address in the BASE field plus four times interrupt cause number.

vogelpi, 2019-07-03:
The changes discussed here were implemented and merged in PR #104 .
Regarding `mtvec`: It can be modified by changing the boot address at runtime.

However, this will change in the future. We will at some point make `mtvec` writable by software. This will allow to change it independently of the boot address. This work is tracked in issue #118

closed 2019-07-03 by vogelpi