lowRISC/ibex #72: Reference integration for FPGA

imphil, 2019-06-07:
Goal: Provide a minimal, but fully working example of how to use Ibex on a well-known FPGA board.

Target board: One of the well-known Xilinx 7-series boards, Nexys or Arty*.

This will require:
- A toplevel
- Additional hardware blocks
  - A RAM
  - A prim_clock_gating implementation
  - A trivial GPIO IP would be nice which maps some bus reads/writes to LEDs/Switches.
- A board-dependent constraints file (xdc)
- Build system support (a fusesoc core file)
- Documentation and code to run a minimal hello world (including linker file, crt)

I'm not sure about where to put these things into the repository, I'd most likely go for a top-level directory like `examples/fpga/nexyvideo` and `examples/sw/hello_world`.

imphil, 2019-06-07:
Talk to me when you're getting started on this issue, I have some existing code to share that we can reuse.

towoe, 2019-07-15:
Implemented with PR https://github.com/lowRISC/ibex/pull/78

closed 2019-07-15 by towoe