lowRISC/ibex #80: Number of performance counters

gautschimi, 2019-06-19:
the parameter MHPMCounterNum does not seem to influence the number of performance counters. I synthesized the core once with MHPMCounterNum=1 and then with MHPMCounterNum=8 and it results in almost exactly the same area.

by looking at the code I also don't see how this parameter is changing the number of physically present counters/flipflops. I'm pretty sure there are 32 64b registers and 32 64b adders no matter what the parameters are.

I think MHPMCounterWidth has the same problem. it probably only counts to the desired width, but the registers [63:MHPMCounteWidth] cannot be optimized away.

the problem with the width can be easily fixed by changing:
```
csr_rdata_int = mhpmcounter_q[mhpmcounter_idx][63:32];
```
to
```
csr_rdata_int[MHPMCounterWidth-1:0] = mhpmcounter_q[mhpmcounter_idx][MHPMCounterWidth-1:32];
csr_rdata_int[32-1:MHPMCounterWidth-32] = '0;
```

if mhpmcounter_q[63:MHPMCounterWidth] is never accessed, the synthesizer can optimize it away. otherwise these registers are there but cannot really be used.

vogelpi, 2019-06-19:
Thanks @gautschimi for reporting this, I will look into it.

gautschimi, 2019-06-20:
Ok what I wrote was not 100% correct. I think the counters 11-32 were never generated.

however, there is still something wrong. the number of registers is the same for MHPMCounterNum=0,1,2,3,4,5,6,7,8

I think apart from the cycle counter and instruction counter no other register is generated. but it might also be my synthesis tool, it reports quite a few errors when comparing csr_addr_i to CSR_MCOUNTER_MASK (which contains X) it might just optimize all other counters away.

this can be solved by using unique case**x**
e.g.:

```
unique casex (csr_addr_i)
CSR_MCOUNTER_MASK:
csr_rdata_int = mhpmcounter_q[mhpmcounter_idx][31: 0];
```
you can try my patch:

[mhpm_patch.txt](https://github.com/lowRISC/ibex/files/3309861/mhpm_patch.txt)

vogelpi, 2019-06-20:
Hi @gautschimi ,
thanks for reporting back. I checked with Vivado. It optimizes all counters away during synthesis except for mcycle and minstret. I am not 100% sure but I think this happens because of the `X` in the `CSR_MCOUNTER_MASK` etc. We are not using `casex` according to our [Verilog style guide](https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md). I need to think about a different solution.

What synthesis tool are you using?

gautschimi, 2019-06-20:
cadence rtl compiler

ok I didn't know about your guidelines. I think casex is useful if used with care. but if you don't allow casex you should also not allow if else statements with 'x' inside. 

to get rid of it you can just do
```
case (csr_addr_i[11:5])
```

vogelpi, 2019-06-20:
I have worked out a solution to get rid of the `X` in the mask and to let the synthesizer optimize away unused counters. It works in Vivado. @gautschimi Could you please also test Cadence RTL compiler?

The fix is here: e8615301b25c5e1c512aaa4eadf7426884cddec6

gautschimi, 2019-06-21:
I can confirm that it also works with rtl compiler. thanks a lot!

vogelpi, 2019-06-21:
Thanks for reporting back, pal. Btw, you also need the fix in #84 to properly work with the counters.

gautschimi, 2019-06-21:
ah there is still one issue:
when MHPMCounterNum < 8
my simulator complains because:
```
ncelab: *E,MULDRN (/home/mgautschi/design/EMBT/digital/ip/ibex/rtl/ibex_cs_registers.sv,519|30): Variable 'tb.top_i.i_core_region.i_core.cs_registers_i.mhpmcounter_incr' has multiple conflicting drivers and the indicated continuous assignment in scope 'tb.top_i.i_core_region.i_core.cs_registers_i' is contributing to this conflict set.
  assign mhpmcounter_incr[4]  = imiss_i & ~pc_set_i; // cycles waiting for instr fetches ex.
                              |
ncelab: *E,MULDRN (/home/mgautschi/design/EMBT/digital/ip/ibex/rtl/ibex_cs_registers.sv,520|30): Variable 'tb.top_i.i_core_region.i_core.cs_registers_i.mhpmcounter_incr' has multiple conflicting drivers and the indicated continuous assignment in scope 'tb.top_i.i_core_region.i_core.cs_registers_i' is contributing to this conflict set.
  assign mhpmcounter_incr[5]  = mem_load_i;          // num of loads
                              |
ncelab: *E,MULDRN (/home/mgautschi/design/EMBT/digital/ip/ibex/rtl/ibex_cs_registers.sv,522|30): Variable 'tb.top_i.i_core_region.i_core.cs_registers_i.mhpmcounter_incr' has multiple conflicting drivers and the indicated continuous assignment in scope 'tb.top_i.i_core_region.i_core.cs_registers_i' is contributing to this conflict set.
  assign mhpmcounter_incr[6]  = mem_store_i;         // num of stores
                              |
ncelab: *E,MULDRN (/home/mgautschi/design/EMBT/digital/ip/ibex/rtl/ibex_cs_registers.sv,523|30): Variable 'tb.top_i.i_core_region.i_core.cs_registers_i.mhpmcounter_incr' has multiple conflicting drivers and the indicated continuous assignment in scope 'tb.top_i.i_core_region.i_core.cs_registers_i' is contributing to this conflict set.
  assign mhpmcounter_incr[7]  = jump_i;              // num of jumps (unconditional)
                              |
ncelab: *E,MULDRN (/home/mgautschi/design/EMBT/digital/ip/ibex/rtl/ibex_cs_registers.sv,524|30): Variable 'tb.top_i.i_core_region.i_core.cs_registers_i.mhpmcounter_incr' has multiple conflicting drivers and the indicated continuous assignment in scope 'tb.top_i.i_core_region.i_core.cs_registers_i' is contributing to this conflict set.
  assign mhpmcounter_incr[8]  = branch_i;            // num of branches (conditional)
                              |
ncelab: *E,MULDRN (/home/mgautschi/design/EMBT/digital/ip/ibex/rtl/ibex_cs_registers.sv,525|30): Variable 'tb.top_i.i_core_region.i_core.cs_registers_i.mhpmcounter_incr' has multiple conflicting drivers and the indicated continuous assignment in scope 'tb.top_i.i_core_region.i_core.cs_registers_i' is contributing to this conflict set.
  assign mhpmcounter_incr[9]  = branch_taken_i;      // num of taken branches (conditional)
                              |
ncelab: *E,MULDRN (/home/mgautschi/design/EMBT/digital/ip/ibex/rtl/ibex_cs_registers.sv,526|30): Variable 'tb.top_i.i_core_region.i_core.cs_registers_i.mhpmcounter_incr' has multiple conflicting drivers and the indicated continuous assignment in scope 'tb.top_i.i_core_region.i_core.cs_registers_i' is contributing to this conflict set.
  assign mhpmcounter_incr[10] = is_compressed_i      // num of compressed instr
                              |
ncelab: *E,MULDRN (/home/mgautschi/design/EMBT/digital/ip/ibex/rtl/ibex_cs_registers.sv,527|30): Variable 'tb.top_i.i_core_region.i_core.cs_registers_i.mhpmcounter_incr' has multiple conflicting drivers and the indicated continuous assignment in scope 'tb.top_i.i_core_region.i_core.cs_registers_i' is contributing to this conflict set.
    assign mhpmcounter_incr[i] = 1'b0;
                               |
ncelab: *E,MULDRN (/home/mgautschi/design/EMBT/digital/ip/ibex/rtl/ibex_cs_registers.sv,531|31): Variable 'tb.top_i.i_core_region.i_core.cs_registers_i.mhpmcounter_incr' has multiple conflicting drivers and the indicated continuous assignment in scope 'tb.top_i.i_core_region.i_core.cs_registers_i.gen_mhpmcounter_incr_inactive[3]' is contributing to this conflict set.
    assign mhpmcounter_incr[i] = 1'b0;
                               |
ncelab: *E,MULDRN (/home/mgautschi/design/EMBT/digital/ip/ibex/rtl/ibex_cs_registers.sv,531|31): Variable 'tb.top_i.i_core_region.i_core.cs_registers_i.mhpmcounter_incr' has multiple conflicting drivers and the indicated continuous assignment in scope 'tb.top_i.i_core_region.i_core.cs_registers_i.gen_mhpmcounter_incr_inactive[4]' is contributing to this conflict set.
    assign mhpmcounter_incr[i] = 1'b0;
                               |
ncelab: *E,MULDRN (/home/mgautschi/design/EMBT/digital/ip/ibex/rtl/ibex_cs_registers.sv,531|31): Variable 'tb.top_i.i_core_region.i_core.cs_registers_i.mhpmcounter_incr' has multiple conflicting drivers and the indicated continuous assignment in scope 'tb.top_i.i_core_region.i_core.cs_registers_i.gen_mhpmcounter_incr_inactive[5]' is contributing to this conflict set.
    assign mhpmcounter_incr[i] = 1'b0;
                               |
ncelab: *E,MULDRN (/home/mgautschi/design/EMBT/digital/ip/ibex/rtl/ibex_cs_registers.sv,531|31): Variable 'tb.top_i.i_core_region.i_core.cs_registers_i.mhpmcounter_incr' has multiple conflicting drivers and the indicated continuous assignment in scope 'tb.top_i.i_core_region.i_core.cs_registers_i.gen_mhpmcounter_incr_inactive[6]' is contributing to this conflict set.
    assign mhpmcounter_incr[i] = 1'b0;
                               |
ncelab: *E,MULDRN (/home/mgautschi/design/EMBT/digital/ip/ibex/rtl/ibex_cs_registers.sv,531|31): Variable 'tb.top_i.i_core_region.i_core.cs_registers_i.mhpmcounter_incr' has multiple conflicting drivers and the indicated continuous assignment in scope 'tb.top_i.i_core_region.i_core.cs_registers_i.gen_mhpmcounter_incr_inactive[7]' is contributing to this conflict set.
    assign mhpmcounter_incr[i] = 1'b0;
```
you should use a combinational process to drive these signals, not assign statements.

vogelpi, 2019-06-21:
> you should use a combinational process to drive these signals, not assign statements.

Thanks, good catch. I have fixed this as well.

gautschimi, 2019-06-21:
(y) thanks!

PR #83 (2019-06-24): Fix perf counters
This PR solves two problems in the performance counters:
1. Fix increment of performance counters
2. Make sure performance counters are not accidentally optimized away during synthesis. Only those are optimized away, which are disabled.

It resolves #80.

The changes have been tested using Verilator and Vivado.
closed 2019-06-24 by imphil