lowRISC/ibex #87: Test trap precision

imphil, 2019-06-24:
Add DV to ensure that traps are as precise as we expect them to be. This requires the expectations to be documented first in  #86.

imphil, 2019-06-24:
@taoliug is this something the riscv-dv can do already?

taoliug, 2019-07-11:
Yes, MSTATUS/MEPC/MTVAL/MCAUSE are checked in the trap handler routine. I think we need to be a bit more specific about MTVAL because the privileged spec allow different implementations for certain exception type.

imphil, 2019-07-29:
@taoliug does your reply imply that MSTATUS/MEPC/MTVAL/MCAUSE are tested to be precise traps, and that these tests currently succeed?

taoliug, 2019-07-29:
It's part of trap handling sequence of the instruction generator. So far illegal instruction/ebreak/ecall trap has been verified. Need to verify other exceptions like load/store access fault etc.