lowRISC/ibex #89: Implement  bit manipulation (B) RISC-V ISA extension

imphil, 2019-06-24:
The bit manipulation extension is currently being specified. Bit manipulation is a common task in many scenarios where Ibex could be used, so we should think about implementing this extension once it has become sufficiently stable.

esshiu, 2019-11-08:
Hi, Philipp,

I'm a new Github user. I'd like to start a project. Can I try to take on this RTL task?

-Eric

imphil, 2019-11-11:
Hi Eric,

thanks for your interest in working on this project! This is a reasonably large project; before we can move on to implementation, we need to do a bit of planning and break it down into smaller actionable chunks. Would you be interested in driving that? (If not that's fine as well, just let us know.)

A potentially good way of moving forward would be a shared Google Doc (share the link in this issue) with an implementation plan, describing 

- which (subsets of) instructions would be good to implement in an Ibex-class microcontroller
- what latency you'd expect for these instructions (single or multiple cycles?)
- a rough first idea of an implementation architecture, e.g. in a block diagram (https://ibex-core.readthedocs.io/en/latest/pipeline_details.html could be a starting point)
- a rough outline of milestones (which parts do you plan to tackle first?)
- your plans for testing during development, and help you need there from the people working on DV

None of that needs to be polished, feel free to share an incomplete draft and ask for help. You can also join our chat channel at https://lowrisc.zulipchat.com/#narrow/stream/198227-ibex (most people are in European time zones).

taoliug, 2019-11-11:
Add one note on the DV side. Right now riscv-dv doesn't support B-extension. As far as I know, the spec is not ratified yet, so it's subject to changes. Like Philipp mentioned, we need to define the scope we want to support. In the meantime, make sure the toolchain(instruction generator, compiler, ISS) is ready.

esshiu, 2019-11-11:
Philipp,

We're interested in the following instructions. We think it's possible to make them single-cycle instructions.

clz (count leading zeros)
fls (find the largest bit set). eg 31 - clz
pcnt (population count = number of 1-bits)
bits (bit-set-by-position) and bitsi (immediate version)
bitc (bit-clear-by-position) and bitci (immediate version)
grev (generic reverse) and grevi (immediate version)

-Eric

sjgitty, 2019-11-11:
IMO we should implement the whole bitmanip extension. Is that not feasible?
Also, @esshiu  please consider the option of doing this with a parameter if it turns out the area impactis large. @GregAC  is researching the area of what is worthy to parameterize (though he is OOO for this and next week)

esshiu, 2019-11-11:
The above instructions(clz, fls, pcnt, bits,bitc, grev,...etc) are high priority per Randall. If we're short on resource or time, we may prioritize these instructions first. I have not studied the complete bit manipulation spec. 

-Eric

esshiu, 2019-11-12:
By the way, it seems "fls","bits, bitsi", "bitc, bitci" are not in the B-extension spec.

-Eric

esshiu, 2019-11-26:
@imphil Friendly ping. Any comments on how to add "fls","bits, bitsi", "bitc, bitci" into the B-extension spec?

imphil, 2019-11-26:
This issue is about implementing the B extension. If you'd like to implement instructions which are not part of the B extension today then there are two ways forward: one is to convince the authors of the B extension to include these instructions. Please follow up with the RISC-V bitmanip working group on that. The other option is adding a non-standard ISA extension to Ibex. That's something we generally try to avoid, but it can be discussed if there's a good use case. Please open a new issue for each such instruction, together with a clear motivation why this is necessary.

Now back to the bitmanip extension: to move this issue forward we need a implementation plan, e.g. in the form of a Google Doc. Please have a look at https://github.com/lowRISC/ibex/issues/89#issuecomment-552406804 and let me know where I can help out.

esshiu, 2019-11-26:
Philip,

We'll likely go with adding a non-standard ISA extension to Ibex. We feel it's a faster route. I will start a new issue for the new instructions.

Thanks.

-Eric

esshiu, 2019-12-02:
Philip,

I started the discussion https://github.com/lowRISC/ibex/issues/495#issuecomment-558996640

The conclusion is that it's okay not to support "fls". "sbset"(aka. "bits") and "sbclr"(aka "bitc") are already in https://github.com/riscv/riscv-bitmanip/blob/master/bitmanip-0.92.pdf.

I'll look into the implementation plan for the following high priority instructions.

clz (count leading zeros)
pcnt (population count = number of 1-bits)
bits (bit-set-by-position) and bitsi (immediate version)
bitc (bit-clear-by-position) and bitci (immediate version)
grev (generic reverse) and grevi (immediate version)

-Eric

asb, 2019-12-03:
Hi Eric. I'd strongly counsel against implementing subsets of the bitmanip extension other than the proposed Zb* subsets. If you feel that the currently proposed Zb* subsets don't match your needs, now is the time to suggest alternatives to the RISC-V bitmanip working group. The huge number of possible RISC-V has advantages, but can of course also be a burden in terms of compiler support and testing.

In your case, it looks like Zbb, Zbs, and Zbp cover the instructions you need.

esshiu, 2019-12-03:
Alex,

Yes, you're right. Zbb, Zbs, and Zbp cover our need.

-Eric

vogelpi, 2019-12-16:
Adding @ganoam to the discussion. He is joining us in January and as part of his Master thesis do a design space exploration on the current draft of the bit-manipulation extension. I.e., he will also look at other subsets of the spec and how to implement it in Ibex.

imphil, 2019-12-17:
I went ahead and added an implementation plan and an implementation tracking sheet to the description of this bug. Please have a look at these documents for more information.

We'll use this bug for tracking the overall progress towards implementing the bitmanip extension. Please use individual issues and PRs to work on parts of the problem.

To get started, I've opened #526 to implement the first (simple) instruction, together with the initial framework (parameter, documentation skeleton, etc.).

GregAC, 2019-12-17:
Thanks for setting up the documents @imphil.

Looking through the tracking spreadsheet not all parts of the bitmanip spec are in there, Zbc, Zbr, Zbm appear to be missing.

ganoam, 2019-12-17:
Hi Guys, 
Thanks @vogelpi for the introduction and thank you @imphil  for preparing the documents! I will start with a few FPGA Optimizations (Issue #522) in January and then move on to this one. 
I am looking forward to working with you. 
see you soon,
Noam

imphil, 2019-12-17:
Good point Greg, Zbc and Zbr were missing, Zbm is RV64-only.

esshiu, 2019-12-17:
@imphil Can you please share the link to the implementation plan and an implementation tracking sheet?

Thanks.

-Eric

imphil, 2019-12-17:
Eric, please look at the top of this issue, they see both linked there.

ganoam, 2020-03-09:
Hi Guys,

I am finally starting with this issue. 

@imphil could you please give me write access to the tracking sheet?

For clarity, I think it would be more suitable to open separate issues not for each individual instruction but for goups of instructions, such as {`clz` + `ctz` + `pcnt`}, {`slo` + `sro`} and so on. Much of the hardware will be shared, between those instructions so it would not make much sense to discuss them separately.

My plan, if everyone is ok with it, is to start with single-cycle versions of every instruction. Once we have this we should decide where to optimize for area and timing. As for evaluating the usefulness of individual instructions using benchmark applications, the results may be of limited significance as the GCC backend code generator as of yet only supports a subset of the B extension. But I think we can discuss that at a later time.

vogelpi, 2020-03-09:
This sounds reasonable to me @ganoam . I would do the instructions as groups and everything single-cycle as long as the area/timing does not "explode". If things change, I would analyze the critical instruction in isolation (using a separate issue).

ganoam, 2020-03-09:
Ok, I have opened issues for subsets of the `zbb` (base) group of the B extension:
- #526: Bit Counting: `clz`, `ctz and `pcnt`(@imphil  could you update the title to include the other two instructions?)
- #663: Shifting, Rotating, Reversing and Orcombine: Quite a large group with lots of potential to share resources: `slo[i]`, `sro[i], `rol`, `ror[i]`, `rev8`, `rev`, `orc.b`.
- #665: Comparison: `min`, `max`
- #667: Logic-with negate: `andn`, `orn` and `xnor`.
- #668: Pack two words in one register: `pack[h][u]`.
These instructions constitute `zbb` base group as defined in the draft specification. I will prepare a draft PR, so we can discuss my code proposals more directly. Also I will do some synthesis runs to see the impact on area and timing.

ganoam, 2020-03-11:
A PR is available for implementing the zbb extension #683. It contains all the changes discussed in the above mentioned issues. 

I have made this PR in order to have a central place to discuss the code. If you prefer, I can also split it up and do one for each group of instructions.

vogelpi, 2020-03-29:
Update: Support for Zbb is now there and can be enabled with a new parameter RV32B (still experimental). A second PR adding support for Zbt got published by @ganoam recently: #746

ganoam, 2020-04-28:
Update: Zbt (#746) and Zbs (#799) extensions have been merged. A PR adding Zbp is ready for review: #821

ganoam, 2020-05-12:
A PR for Zbe is ready for review: #863.

ganoam, 2020-05-22:
Update: We are done implementing all the instructions, once #892 has been merged. 

The next steps will be: 
Do a final round of optimization: 
- It has been observed, that there is a large discrepancy between the area consumption when the design is synthesized with relaxed timing constraints vs. tight timing constraints: The Implementation of the whole RVB extension grows the design by 6.0 kGE (25 / 24%) when synthesized with relaxed timing constraints (both 2-stage and 3-stage configurations) and 9.2 / 14.0 kGE (45 / 28%) respectively with tight timing constraints. 
Enabling RVB instructions increases the overall critical path by ~0.1ns. The approach will be to move RVB operations out of the critical path wherever possible. This should mitigate both the area increase and the impact on timing. 

- Parameterize the implementation:
It has been requested to deliver to versions of the RVB extension: Once a balanced implementation with a reasonable footprint, featuring probably just Zbb, Zbs, Zbf and Zbt, and on the other hand a high performance version.
Here I will need your input: How would you like me to set up this configuration? 
The manner in which the multiplier unit is parameterized to select the Implementation (`parameter bit RVM` + `parameter string MultiplierImplementation`) would have been my model. But it has been a subject of discussion previously. 
Some possibilities that come to mind: 
- `RVB = (0=no, 1=balanced, 2=performance)` 
- Two Parameters `RVB_BALANCED`, `RVB_PERFORMANCE`

imphil, 2020-06-09:
> Ideally, we could use an enum but I don't know if that is possible in SystemVerilog.

It is and we use that in OpenTitan at least for the `Impl` parameter to our primitives. We also pass that parameter to the tool through a command line (a define in that case, https://github.com/lowRISC/opentitan/blob/master/hw/top_earlgrey/top_earlgrey_verilator.core#L65). I initially thought it would be rather fragile, but seems to work well so far with all tools we've tried.

GregAC, 2020-06-09:
Using a enum with a define to choose it sounds reasonable to me. We already need to use a define to choose which string to pass for VCS anyway (as it doesn't support passing string parameters on the command line) so the infrastructure around using ibex_config.py to do this already exists.

imphil, 2020-06-09:
To clarify: if you only want a parameter, you shouldn't need a define. The example I linked to from OpenTitan uses a define for entirely different reasons.

GregAC, 2020-06-09:
I suspect VCS will require a define to pass the enum setting through from the command line but I may be surprised. In any case that can be isolated to the UVM DV environment (as it is now with the pass string via define method).

ganoam, 2020-06-10:
Perfect, thanks guys. I'll do it with the enum then.

vogelpi, 2020-07-03:
The final PR from @ganoam has gone in last week (#965) and the implementation can successfully be verified using RISCV-DV. Thanks for the awesome work Noam!

I have now created a PR to clarify the situation around the draft status here: #1011 

Other than that, I think we are done here and I suggest to close the issue WDYT @imphil ?

closed 2020-07-06 by vogelpi