lowRISC/ibex #90: Implement sleep with wfi instruction

imphil, 2019-06-24:
We currently implement the WFI instruction as NOP. This implementation is correct, but we should think about putting the core to sleep instead to save power.

tjaychen, 2019-06-24:
Can you comment a bit on how WFI=NOP works?
Does this mean the user is expected to do something like the following?

``` 
while (some_interrupt_condition) {
     wfi;
   } 
```
In the situation, will ibex continually fetch code? I'm guessing not, we'll just in some tight NOP loop?

davideschiavone, 2019-06-25:
Hi @imphil , actually the WFI instrunction is not implemented as a NOP.
it puts the core into a SLEEP state (https://github.com/lowRISC/ibex/blob/master/rtl/ibex_controller.sv#L225) where the clock enable is deasserted. Have a look at https://github.com/lowRISC/ibex/blob/master/rtl/ibex_core.sv#L257

Best
Davide

imphil, 2019-06-25:
Thanks @davideschiavone for jumping in here, I got pretty much confused by wfi in debug mode (where it is a nop) and didn't double-check the normal case before opening this issue. 

But that now gives me the joy of closing this issue as "done" without having to do any implementation work :smile:

closed 2019-06-25 by imphil