lowRISC/ibex #91: We need to document working versions of CAD tools

jrrk, 2019-06-25:
The Google UVM framework requires either VCS or incisive.
I have found that vcs-mx/N-2017.12-SP1 works but that vcs-2016.06-sp1 does not.

We should record these experiences in some sort of compatibility table.

The failing version crashes out, with a fatal error.

Fzl1029, 2020-03-19:
Hello! I encountered Error as "System functions calls are not yet implemented in constraints." when compiling google_riscv-dv/riscv_vector_cfg.sv. Is this error caused by software version? I use vcs-mx2015. Thank you!

imphil, 2020-03-26:
It's hard to say, but please give it a try with the latest VCS version.

rswarbrick, 2021-01-27:
There's some work afoot to add this to the `tool_requirements.py` code, on the OpenTitan side. Once that work has been vendored in (see also #985), we should be able to do this pretty trivially.

PR #1324 (2021-04-06): [util] Document minimum VCS version
Fixes #91.
closed 2021-04-06 by rswarbrick