lowRISC/ibex #94: interrupt & interrupt_disable

gautschimi, 2019-06-25:
the core wakes up when an interrupt is received, but the mie=0. I think that's a bug.

https://github.com/lowRISC/ibex/blob/master/rtl/ibex_core.sv#L279

should be:
```
  assign clock_en    = core_busy |  (irq_i & m_irq_enable)  | debug_req_i;
```

I'm also not sure about the behaviour of mret. I think if mret is executed twice it will enable the mie which could lead to problems:
https://github.com/lowRISC/ibex/blob/master/rtl/ibex_cs_registers.sv#L418-L426

in ibex_cs_registers.sv:
```
      csr_restore_mret_i: begin //MRET
        mstatus_n.mie  = mstatus_q.mpie;
        mstatus_n.mpie = 1'b1;
      end //csr_restore_mret_i

      csr_restore_dret_i: begin //DRET
        mstatus_n.mie  = mstatus_q.mpie;
        mstatus_n.mpie = 1'b1;
      end //csr_restore_dret_i
```
I think mpie should be reset to either 0 or the value of mie, but I'm not sure what the correct behaviour is:
```
      csr_restore_mret_i: begin //MRET
        mstatus_n.mie  = mstatus_q.mpie;
        mstatus_n.mpie = mstatus_q.mie;
      end //csr_restore_mret_i

      csr_restore_dret_i: begin //DRET
        mstatus_n.mie  = mstatus_q.mpie;
        mstatus_n.mpie = mstatus_q.mie;
      end //csr_restore_dret_i
```

davideschiavone, 2019-06-25:
Priv spec page 41:

The WFI instruction can also be executed when interrupts are disabled. The operation of WFI must
be unaffected by the global interrupt bits in mstatus (MIE/SIE/UIE) and the delegation registers
[m|s]ideleg [...]

Regarding the MRET, what do you mean can cause problem? This is how MRET is defined in section "5.9 Trap Return" of the priv spec.

gautschimi, 2019-06-25:
oh yes you are right regarding the wfi. thanks!

and also mret seems to be according to specs. but it's still a bit weird to me.

I just had a test in which mret was executed twice in a row (not on purpose) and this enabled my interrupts. because the first execution is setting mpie=1 and the second is then setting mie=mpie=1. => interrupts enabled.
anyway not so important

davideschiavone, 2019-06-25:
The WFI makes sense otherwise you can end up in a deadlock :)
Regarding the MRET, that's true, it's a bit weird :)

closed 2019-06-26 by gautschimi