lowRISC/ibex #96: Boot Address

eroom1966, 2019-06-26:
Hi
Something strange is happening, I have defined the Reset / Boot address to be 0x80000000
but the first address fetched is 0x80000080
How do I define the reset address I wish to be the first fetch address ?

Thx
Lee

gautschimi, 2019-06-26:
the reset vector is always at boot_addr + 0x80. the addresses up to 0x80 are used for the interrupt vector table

imphil, 2019-06-26:
We'll have in our tree shortly, but for now you can find a suitable linker script and crt in this PR: https://github.com/lowRISC/ibex/pull/78/files

eroom1966, 2019-06-26:
How does this correspond to the RISCV Priveleged spec ?
I see the exceptions are at hard coded addresses, rather than being encoded into an MCAUSE register
https://ibex-core.readthedocs.io/en/latest/exception_interrupts.html

Thx
Lee

imphil, 2019-06-26:
The current trap handling is free-form, which was allowed in an older version of the spec. Work is underway to update it to to the current spec in https://github.com/lowRISC/ibex/issues/70.

vogelpi, 2019-07-03:
The trap handling is now conforming to the RISC-V spec. I am thus closing this issue.

closed 2019-07-03 by vogelpi

jrrk2, 2020-03-12:
What if the boot_addr_i is non-zero ? It seems a bit strange to force the address to be something other than what you asked for.