lowRISC/ibex #103: Diagnose/remove combinatorial loop

imphil, 2019-06-27:
Verilator reports this loop:

```
%Warning-UNOPTFLAT: ../src/lowrisc_ibex_ibex_0.1/rtl/ibex_core.sv:129: Signal unoptimizable: Feedback to clock or circular logic: ibex_core.is_decoding
                    ../src/lowrisc_ibex_ibex_0.1/rtl/ibex_core.sv:129:      Example path: ibex_core.is_decoding
                    ../src/lowrisc_ibex_ibex_0.1/rtl/ibex_controller.sv:527:      Example path: ASSIGNW
                    ../src/lowrisc_ibex_ibex_0.1/rtl/ibex_id_stage.sv:158:      Example path: ibex_core.id_stage_i.deassert_we
                    ../src/lowrisc_ibex_ibex_0.1/rtl/ibex_decoder.sv:550:      Example path: ASSIGNW
                    ../src/lowrisc_ibex_ibex_0.1/rtl/ibex_id_stage.sv:174:      Example path: ibex_core.id_stage_i.jump_in_id
                    ../src/lowrisc_ibex_ibex_0.1/rtl/ibex_id_stage.sv:602:      Example path: ALWAYS
                    ../src/lowrisc_ibex_ibex_0.1/rtl/ibex_id_stage.sv:172:      Example path: ibex_core.id_stage_i.jump_set
                    ../src/lowrisc_ibex_ibex_0.1/rtl/ibex_controller.sv:151:      Example path: ALWAYS
                    ../src/lowrisc_ibex_ibex_0.1/rtl/ibex_core.sv:129:      Example path: ibex_core.is_decoding
```

We need to diagnose this loop in more detail, and hopefully remove it. Otherwise add an explanation and a lint waiver.

(My gut feeling is that we need to apply deassert_we later to signals that need it.)

vogelpi, 2019-06-28:
I had a look, the combinational loop warning goes away, if `jump_in_id_o` and the various other signals at the bottom of `ibex_decoder.sv` are changed not to depend on `deassert_we_i`. To me, it makes little sense to have this kind of control logic inside the decoder. Also, looking at the FSMs I think they don't need to depend on this signal anyway. It seems that for some cases the same is achieved already by the ID-EX/WB FSM in `ibex_id_stage.sv`.

I will thus move these signals out of the decoder and see how to integrate the functionality in the ID stage to simplify and cleanup the whole thing.

vogelpi, 2019-07-03:
The fix for this issue is part of PR #120 .

PR #120 (2019-07-09): Cleanup ID stage and decoder
This PR contains a major cleanup of the ID stage and the decoder. Control logic (e.g. for stalls) that depends not on the decoder exclusively, is moved out of the decoder. Similarly, logic inside the ID stage that is strictly related to instruction decoding (immediate & register file address generation and checking) is moved into the decoder. 

As a result, the ID stage now instantiates the modules of the controller, decoder and registers, and contains the the main muxes, the write-back (WB) FSM to handle multicycle instructions and minimal glue logic, only.
This w [... cut]
closed 2019-07-09 by vogelpi