lowRISC/ibex #105: Instructions accessing x16-x31 should be considered illegal for RV32E

asb, 2019-06-27:
The specification doesn't require that all unused instruction encodings generate illegal instruction traps. However, we prefer to do so for Ibex as this increases the chances of catching errors early and provides some forward-compatibility for trap+emulate of future standard or non-standard extensions.

Right now, an illegal instruction exception won't be signalled on an RV32E system for instructions accessing x16-x31. There is some relevant commented out logic in ibex_id_stage.sv: https://github.com/lowRISC/ibex/blob/90796e3800207c67f5f0349b2216ef8bdf623ab2/rtl/ibex_id_stage.sv#L274 Logic may also be needed in ibex_compressed_decoder.sv for the CI and CSS formats (ultimately, after #104 we need to ensure the mcause holds the compressed version of the instruction).

vogelpi, 2019-07-03:
I have implemented this functionality. But since it is based on PR #120 which first needs to be reviewed and merged. I did not yet create a PR for this.

One thing worth to mention is that the check previously commented out can generate false alarms as the bits used to encode the register file addresses for some instructions can also be used to encode other arguments and/or immediates for other instructions. Thus, the check must also evaluate if the current instruction uses the register file output or not.

As for the compresses instructions, the controller always writes the actual faulting instruction to `mtval`. Independently whether the instruction is detected as being illegal in the decoder or the compressed decoder. This means there is no check of the register file address required in the compressed decoder. I added a comment to the compressed decoder regarding this.

vogelpi, 2019-07-09:
I am closing this issue. The discussed feature got added with #126 .

closed 2019-07-09 by vogelpi