lowRISC/ibex #108: Interrupt context saving not working

arghimat, 2019-07-01:
The problem is not systematic.
The problem is detected by comparing the files generated by the tracer.sv module, for the zero_riscy and the Ibex.
A small modification (around the is_decoding_o signal)of ibex_controller.sv module permits to avoid the problem (but this modification is not safe to 100 %;   The problem must be present in Debug mode : not tested).
list of file:
zero_riscy_VS_ibex_problem.log    : comparaison, line by line of trace_core.log (generated by trace.sv)
modifications_ibex_controller.log     :  modification that improves the behavior of Ibex during an interruption
ibex_writing_during_an_interruption.png  : waveforms to show the problem
zero_riscy_writing_during_an_interruption.pgn   : waveforms of zero_riscy (no problem)
[zero_riscy_VS_ibex_problem.log](https://github.com/lowRISC/ibex/files/3320582/zero_riscy_VS_ibex_problem.log)
[modifications_ibex_controller.log](https://github.com/lowRISC/ibex/files/3320584/modifications_ibex_controller.log)
[image]

[image]
 
[image]
[image]

[modifications_ibex_controller.log](https://github.com/lowRISC/ibex/files/3344870/modifications_ibex_controller.log)
[zero_riscy_VS_ibex_problem.log](https://github.com/lowRISC/ibex/files/3344871/zero_riscy_VS_ibex_problem.log)

vogelpi, 2019-07-01:
Hi @arghimat ,
thanks for creating this issue and providing the details.

To me it looks like that the interplay between the write-back and controller FSMs is not working properly, especially for multi-cycle instructions. I am currently in the process of reworking the interrupt framework inside Ibex. For that, I will also have to have a close look at the two FSMs and your case here.

I will get back to you. One question that just came to my mind: Which register file are you using for Ibex? Because we had an issue recently (#75), where someone used the latch-based register file with an incompatible clock gating cell. This led to register writes not being handled correctly.

Best regards,
Pirmin

jeanlouismaquet, 2019-07-01:
Hello,

Thank you for your comments.
Concerning the module ibex_registers, we use the Flipflops file.

Adding the following line seems to improve the behaviour, but not completely:
          irq_req_ctrl_i && irq_enable_int &&  !debug_req_i && !debug_mode_q: begin
            // Serve an interrupt (not in debug mode)
            ctrl_fsm_ns = IRQ_TAKEN;
            **_is_decoding_o = 1'b1;_**           <=========   added line
            halt_if_o   = 1'b1;
            halt_id_o   = 1'b1;
          end

The problem must appear  in debug mode (???).

I created a waveforms file (in WLF format) and its  .do file. 
I added all FSMs.
Remark: the signals are creatd after +/- 10 ms (zoom the end of waveform).
I added the file created by tracer module (see the cycles signal in waveform and in log file).

I hope this will help you,

Best regards,

Jean-Louis
[trace_problem_irq.zip](https://github.com/lowRISC/ibex/files/3345778/trace_problem_irq.zip)

towoe, 2019-07-02:
Hello,

I changed the input signals on which the tracer relies in PR https://github.com/lowRISC/ibex/pull/77. It does not check for the `is_decoding_o` signal anymore, but just relies on `rvfi_valid`. If you would want to test your trace with the changes from the PR, I would be happy to hear if the tracer is still working for you and/or if this problem is still occurring.

Thanks,
Tobias

jeanlouismaquet, 2019-07-03:
Hello Tobias,
The Tracer functionality isn't the problem. I use it to debug.
The problem is in the FSMs under ibex_id_stages (ibex_controller).
Sometimes, when an interrupt occurs, the last write in registers is not present.
I send you some pictures of my simulation (modelsim) and  its wlf file (if you have modelsim).
During the conversion wlf2vcd, the FSMs signals dissapear. I can not send you a correct vcd file.
I also provide a picture of the same simulation of the zero_riscy (Write in x14 register is OK).
I send you a file (generated with the tracer module) showing the difference between the zero_riscy and Ibex.
Sent files (in zip file):
ibex_problem_3.png
ibex_problem_2.png
ibex_problem_1.png
waves_E521_ibex_new.wlf
zero_riscy_VS_ibex.png
zero_riscy_write to_register_during an interrupt(.png

Best regards,

Jean-Louis
[To_lowRISC_issue03July2019.zip](https://github.com/lowRISC/ibex/files/3353861/To_lowRISC_issue03July2019.zip)

vogelpi, 2019-07-04:
Hi @jeanlouismaquet , hi @arghimat ,

thanks for sending the additional information. I just wanted to let you know that I did not forget about your problem. There is a problem in the controller and how it prioritizes interrupts, exceptions and normal instructions. In your particular case, the ongoing load is interrupted by the interrupt. The controller jumps into another state and deasserts `is_decoding`. This deasserts the write enable for the register file, meaning the value returned by the load from memory is lost.

I had to do some other work on the decoder and ID stage previously, but I will now focus on this problem here. I will let you know if I have news.

Kind regards,
Pirmin

vogelpi, 2019-07-05:
Hi @jeanlouismaquet , hi @arghimat ,

I have reworked the faulty controller FSM to solve the problem you have reported. However, this fix depends on other changes currently under review here #120 . Thus I did not create a PR for these latest changes.

Nevertheless, the changes are available on my development branch here: https://github.com/vogelpi/ibex/tree/vogelpi-dev

Would you mind to checkout this branch and give me some feedback if it properly works in your setup?
Best regards and have a nice weekend,
Pirmin

commit 9ef3c13f (2019-07-11): Fix control for interrupts, debug request and multicycle instr

This commit makes sure that interrupts and debug requests do not
interrupt currently running multicycle instructions. Priority is
given 1) to the currently running instruction, possible exceptions
caused by this instruction, 2) debug requests, 3) interrupt
requests.

Previously, currently running instructions were aborted upon
incoming debug and interrupt requests, which could corrupt the
processor state and lead to exceptions being ignored.

This commit resolves #108 and #121.
closed 2019-07-11 by vogelpi