lowRISC/ibex #114: Data memory interface

ivanmgribeiro, 2019-07-01:
Hi,
I'm trying to use the ibex core with an Avalon memory slave, and I can't find a clean mapping between the ibex memory interface signals and Avalon's signals. I have a couple of queries about the data memory interface:

- From my understanding of it, it is possible to make a request, have it granted, and then make another request before the first one has been responded to. Is it then possible for this second request to be granted before the first one has been responded to? If so, when there is a response, how do you know which request is being responded to?
- Is there a particular reason why there is an assertion checking that rvalid should not be set while the Load-Store unit is idle? The core seems to work both in simulation and on an FPGA when the value of rvalid is set combinationally, but the simulation repeatedly states "Data rvalid set while LSU idle".

For reference, I am using the following assignments to generate the ibex signals from avalon signals:

```
always_ff @(posedge clock) begin
    write_good <= !avm_main_waitrequest && data_req_i && data_we_i;
end

assign data_rvalid_i = write_good || avm_main_readdatavalid;
assign data_gnt_i = !avm_main_waitrequest && data_req_o;
```

Also, if you know a better way of doing this please let me know.

Thanks,
Ivan

towoe, 2019-07-02:
Hello Ivan,

I assume you already looked at the [LSU doc](https://ibex-core.readthedocs.io/en/latest/load_store_unit.html)? And that your assumption for the second request might be answered in the second point of the Protocol section. If a request is made and it is then granted, the core is allowed to change the output signals in the next cycle. But the response for the request has still to be made with a assertion of the `data_rvalid_i` signal. The difference between the grant and valid can be seen quite clearly in Figure 5.

The assertion checking just reports that it sees a asserted valid signal where there should be none. If no request is made, how should there be a valid signal. But I have to admit that in my simple tracer example in PR https://github.com/lowRISC/ibex/pull/77 the valid signal is also asserted all the time, which you should not do.

Tobias

vogelpi, 2019-10-31:
Thanks @towoe for answering this question.

In the meantime, the data interface has been slightly reworked starting with lowRISC/ibex#276 to avoid a dependency of `req_o` on `gnt_i/rvalid_i`. The core can now also output the second transfer without having received the response of the first one. In general, the core expects responses to arrive in the same order as the requests were granted.

I am now closing this issue. Feel free to re-open it if something is still unclear @ivanmgribeiro .

closed 2019-10-31 by vogelpi