lowRISC/ibex #115: ReadTheDocs rendering bug with Chrome

ikarageo, 2019-07-01:
Figure 6 in https://ibex-core.readthedocs.io/en/latest/register_file.html appears minuscule in  Google Chrome 75.0.3770.100 for Windows 10.

towoe, 2019-07-05:
Hello,
I used Chrome (same version) in Windows 10 to check the image. For me there is no problem with Figure 6. So the problem might have another cause and is not directly related to the browser and the OS?

ikarageo, 2019-07-05:
Hi, 
Thanks for checking it. After further inspection I concluded that it is caused by a chrome extension. Normally I check things in incognito which disables all extensions, but I accidentally had this extension enabled also in incognito. Inspect element showed me the following for the `div` the image belongs to:
```
element.style {
}
injected stylesheet
#timing {
    width: 25px;
    margin: 5px;
}
...
```
This shows that the 25px width constraint was injected (by an extension).

imphil, 2019-07-05:
Thanks for checking @ikarageo. I guess we can close this then as something we cannot do anything about it.

closed 2019-07-05 by imphil