lowRISC/ibex #119: ebreak causes PC jump to boot address

taoliug, 2019-07-03:
Please check the screenshot for details. Ideally ebreak should trigger an exception and PC should be changed to the value in MTVAL. However, it seems that ibex jump to boot address instead.

[image]

ibex trace:
             1956000             879 800001cc 4046d693 srai             x13, x13, 0x404     x13=fffffffa x13:ffffffaa
             2010000             906 800001ce 00100073 ebreak                              
             2058000             930 80000000 0800006f jal              x0, 128            
             2118000             960 80000080 0100006f jal              x0, 16             
             2122000             962 80000090 feb58037 lui              x0, 0xfeb58000     

spike trace:
core   0: 0xffffffff800001cc (0x00008691) srai    a3, a3, 4
3 0x800001cc (0x8691) x13 0xfffffffa
core   0: 0xffffffff800001ce (0x00100073) ebreak
core   0: exception trap_breakpoint, epc 0xffffffff800001ce
core   0:           tval 0xffffffff800001ce
core   0: 0x000000008001c000 (0x00007119) addi    sp, sp, -128
3 0x8001c000 (0x7119) x 2 0x80019d9c
core   0: 0xffffffff8001c002 (0x0000c002) sw      zero, 0(sp)

Please let me know if more info is needed.

vogelpi, 2019-07-05:
Hi @taoliug ,

thanks for reporting the issue. I am not fully sure I understand what you mean.

Looking at the spec, I think EBREAK should cause the processor to:
1. generate a breakpoint exception.
2. write the address of the EBREAK instruction to `mepc` CSR.

I cannot see 2 happening in the waves provided but according to the source code, this should happen.
The breakpoint exception (1) causes the processor to jump to the trap-vector base address in `mtvec`. This is set to the boot address. Thus I think what you see is correct.

Best regards,
Pirmin

taoliug, 2019-07-05:
If the MTVEC is hard-coded to boot address, how can SW service this exception? I'd expect MTVEC pointing to some software programmable address which has the trap handling routine.

vogelpi, 2019-07-05:
As discussed here #96 , the first instruction is at boot_addr + 0x80. The addresses up to 0x80 are reserved for the interrupt vector table. Offset 0x0 is used for the exceptions.

taoliug, 2019-07-05:
Thanks, sounds good to me. The spec allows read-only implementation of this register.

closed 2019-07-05 by taoliug