lowRISC/ibex #121: Addr && be not stable during read req

elchanan18, 2019-07-04:
I'm seeing data_addr_o and data_be_o change, which data_req_o is high, and data_gnt_i has not been received.
Is it possible I'm misreading the spec?  (or that there are illegal things I might have in the environment which could cause this?)
Thank you.

[image]

vogelpi, 2019-07-04:
Hi @elchanan18 ,
thanks for reporting this issue.

The address changing in Cycle 5 with request high but without getting a grant does indeed not look good. Would you mind to provide some more details? Ideally a VCD file for the whole processor? This would help me to look into it.

Best regards,
Pirmin

elchanan18, 2019-07-04:
Hi Pirmin,
Thanks for the quick response.
Is it possible to attach a VCD file to this thread?  (about 75K)

vogelpi, 2019-07-04:
Yes, that should work. You can just attach it here.

elchanan18, 2019-07-04:
I had to rename the file type to .txt.
It would let me drag a .vcd

[addr_not_stable.txt](https://github.com/lowRISC/ibex/files/3359295/addr_not_stable.txt)

vogelpi, 2019-07-04:
Thanks for uploading the VCD (to get around the restriction regarding the name and save some space, you can also do a zip).

I had a look on the VCD. The problem is complicated. The core tries to do a non-aligned load. Non-aligned loads are split into two consecutive aligned loads. The first of these two loads fails (the bus returns an error in `data_err_i`). To store the faulting address in the `mtval` CSR, the LSU cannot store the address incoming for the second load. This results in the output address being changed despite not getting a grant. This could be solved by just aborting the LSU operation on the error.

What makes the situation more complicated is that in your case, while the LSU informs the controller (`ibex_controller.sv`) about the load error, the controller is not sensitive to this information as it is in another state (`DBG_TAKEN_ID` instead of `DECODE`). This is part of a bigger problem we currently have in the controller. There are other issues open because of the same thing. I am working on it.

Best regards,
Pirmin

vogelpi, 2019-07-05:
Hi @elchanan18 

I have reworked the faulty controller FSM to solve the second part of the problem you reported. However, this fix depends on other changes currently under review here #120 . Thus I did not create a PR for these latest changes.

Nevertheless, the changes are available on my development branch here: https://github.com/vogelpi/ibex/tree/vogelpi-dev

Would you mind to checkout this branch and give me some feedback if it properly works in your setup?
Best regards and have a nice weekend,
Pirmin

elchanan18, 2019-07-05:
Here's what I'm getting now.

[image]

elchanan18, 2019-07-05:
[addr_stable.zip](https://github.com/lowRISC/ibex/files/3363257/addr_stable.zip)

vogelpi, 2019-07-05:
Thanks for getting back, I obviously had made an error before. Fixed that now. You can retry, be careful though, I did a force push to my dev branch.

elchanan18, 2019-07-05:
I think I got the new code.

[image]
[addr_stable2.zip](https://github.com/lowRISC/ibex/files/3363431/addr_stable2.zip)

vogelpi, 2019-07-05:
Thanks @elchanan18 !
I analyzed the waves. Behavior of the processor looks good now.

1. It jumps into debug mode at the beginning and then reads a fence instruction with bits [14:12] != 000.
2. This is illegal. The core thus raises an exception and jumps into the debug-mode exception handler (at address 0x1A110808).
3. The first instruction in there is a LOAD, which returns an error from the bus.
4. This again raises an exception, the core jumps back to the debug-mode exception handler (again at address 0x1A110808).
5. The first instruction it gets now from memory is a JAL. This is strange. As the IF stage reads at the same address as before but gets a different value from memory. Is this to be expected in your setup?
6. Anyway, it computes the correct target and stores the PC in a register 1.
7. The core then jumps to the computed target.

Is it expected in your test setup that the memory returns a different value in 5 and 3? If not, this could mean we have a new bug in the IF stage. Please let me know.

Best regards,
Pirmin

elchanan18, 2019-07-06:
Hi.  No, I'm using Formal, and not restricting reads to return the same data.
Are you saying that you made another fix, or that what I'm seeing now is correct?
If the latter, is it OK for addr and be to change w/o gnt ?

imphil, 2019-07-06:
>  If the latter, is it OK for addr and be to change w/o gnt ?

yes, please see https://ibex-core.readthedocs.io/en/latest/instruction_fetch.html

"The IF protocol differs from the LSU protocol in that the address can change while the request is valid (instr_req_o is high)."

elchanan18, 2019-07-06:
Right.  For instr it would be OK.  But we're looking at the LSU data interface.

vogelpi, 2019-07-08:
Hi @elchanan18,
yep that is still wrong. I did not see it on Friday sorry. But thanks for getting back. I am working on a fix for this now.

vogelpi, 2019-07-08:
@elchanan18 ,
I fixed the problem. The problem was that while the LSU aborted the first failing misaligned transaction, some internal control logic did not get cleared. This prevented the next misaligned transaction to be recognized as such.

I again force-pushed to my dev branch in case you want to retry execution.
Best,
Pirmin

elchanan18, 2019-07-08:
Hi,

Now I'm getting an issue with byte enables.

[image]
[addr_stable3.zip](https://github.com/lowRISC/ibex/files/3369382/addr_stable3.zip)

vogelpi, 2019-07-08:
Thanks @elchanan18 , also this should be fixed now.

elchanan18, 2019-07-08:
This looks better.  I've been running Formal for over an hour, and it hasn't found a counter-example yet.  (the other failures popped up in a few seconds)

elchanan18, 2019-07-08:
Confirmed.  After 2 hours, Formal says it will never happen.

vogelpi, 2019-07-09:
Great! Thanks a lot @elchanan18 , your help is much appreciated!

I consider this issue solved, but I'll keep the issue open until the bug fixes are merged into master.

commit 8613e880 (2019-07-09): LSU cleanup, abort misaligned ops if first part fails

This commit cleans up the LSU and fixes two bugs:

1. If a misalgned transaction creates an error during the first
   part of the transaction, the second part is not pushed out and
   the transaction is aborted. Previously, the LSU tried to output
   also the second part but did not adhere to the defined
   protocol.

2. Misaligned operations are again working correctly. Previously,
   the generation of the byte enable and the alignment of read
   data was broken as these operations rely on the updated
   address from the AGU to have the s [... cut]
closed 2019-07-09 by vogelpi