lowRISC/ibex #130: Problem when the value of opcode FSM is OPCODE_JALR

jeanlouismaquet, 2019-07-10:
Hello,

With the new file ibex_decoder.sv (5093e0a), when the opcode FSM = OPCODE_JALR, the pc_mux_o FSM stays at PC_BOOT value. Normally, the pc_mux_o FSM must toogle to  PC_JUMP value.
I created a zip file containing:
 ibex_28may19.log  : result from the ibex_tracer module. No problem 
 ibex_last_version.log  : result from the ibex_tracer module : problem  line 43 (jalr)
 waves_AES_CBC_old.wlf  : waveform (modelsim) : pc_mux_o signal : PC_JUMP -> PC_BOOT -> PC_JUMP -> PC_BOOT  
 waves_AES_CBC_new.wlf : waveform (modelsim) : problem  (pc_mux_o : PCPC_JUMP -> PC_BOOT) 
 ibex_new_jalr.png    printscreen of waveform (version with _**problem)**_
 ibex_old_jalr.png      printscreen of waveform (correct version)

Solution : In ibex_decoder.sv, after the line 239, add the following line:
     **_jump_set_o          = 1'b1;_**

Remark: During compilation of ibex_decoder.sv file with modelsim, I obtain the following error:
 **_Error (suppressible): (vlog-12003) /cad/projects/eSecure/jlm4/design/digital/modules/esecure_cpu/macro/IP_ibex/sources/hdl/ibex_decoder.sv(623): Variable 'illegal_insn_o' written by continuous and procedural assignments. See /cad/projects/eSecure/jlm4/design/digital/modules/esecure_cpu/macro/IP_ibex/sources/hdl/ibex_decoder.sv(554)._** 

Solution : At line 553 : change  the **_illegal_insn_o_** signal by **_illegal_insn_**.

Despite the addition of the line  **_jump_set_o          = 1'b1;_** , my simulation still has errors.
I'm debugging this new problem.

Best regards,

Jean-Louis
[problem_ibex_jalr.zip](https://github.com/lowRISC/ibex/files/3377638/problem_ibex_jalr.zip)

vogelpi, 2019-07-10:
Hi @jeanlouismaquet,

thanks a lot for reporting this issues and for providing a fix! I created a PR #131 to merge it into master.

If you find further issues, I would be happy to help you.
Best,
Pirmin

imphil, 2019-07-11:
Thanks for reporting this bug @jeanlouismaquet! Fixed in https://github.com/lowRISC/ibex/commit/8b4e33711847e4ea88cd0f205b8f66f8e09f8bac. Please reopen if the problem is not resolved.

closed 2019-07-11 by imphil