lowRISC/ibex #136: Illegal JALR instruction is executed unexpectedly

taoliug, 2019-07-11:
Here's the related instruction trace.

97598000           48700 8000f7e0 34391c67 INVALID                             
97638000           48720 00002dac f2051ae3 bne              x10, x0, -204       x10:00001900

0x34391c67 is decoded as illegal instruction correctly, as it's a JALR instruction with bit[14:12] != 0. However, looks like ibex doesn't jump to the boot address correctly to do the trap handling. Instead, the JALR instruction is executed and an instruction from 0x2adc is fetched unexpectedly. 

Waveform snapshot:
[image]

please let me know if more information is needed to debug this issue

vogelpi, 2019-07-11:
Hi @taoliug ,
thanks for reporting this bug. There are currently problems with multicycle instructions (jumps, branches, mult/div, loads/stores) in combination with exceptions and interrupt requests as reported in #108  and #121 . The source of the problem is the main controller FSM inside `ibex_controller.sv`. I have been working on this and created a PR #132 to solve these issues. While this PR seems to fix the previous issues, it must first be reviewed before we can merge it into master. This hould happen soon.

In the meantime, if you want you can try out the fixes in PR #132 to see if it also solves your problem. I would be happy for feedback.

Best regards,
Pirmin

vogelpi, 2019-07-12:
PR #132 has been merged. This bug should now be fixed.

closed 2019-07-16 by vogelpi