lowRISC/ibex #138: Opcode name fance

towoe, 2019-07-11:
While working on https://github.com/lowRISC/ibex/issues/134 I just saw that we now have 
https://github.com/lowRISC/ibex/blob/b203a0f7c70be7c7dbdbb8d6abb97671323e2023/rtl/ibex_defines.sv#L33
as opposed to `OPCODE_MISC_MEM`. I understand that we only have `FENCE` and not `FENCE.I` so this is the only time the opcode is used. Just want to check if we want to keep `OPCODE_FENCE`?

asb, 2019-07-11:
Good catch, our general policy is that these enums should match the spec wherever sensible to allow easy cross-referencing and comparison. So OPCODE_MISC_MEM as that major opcode is called in the current spec is more appropriate.

towoe, 2019-07-11:
Thanks @asb for the clarification, I will create a PR.

asb, 2019-07-11:
Very minor: Unless I'm missing some logic behind it, the ordering in that enum seems pretty arbitrary too. I'd personally prefer it if the order reflected the table 25.1 opcode map in the RISC-V ISA manual. Again, this makes it easier to cross-check the two quickly.

towoe, 2019-07-11:
Will reorder the enum, but just to make it clear, it should be sorted by keeping the higher bits stable? So row after row.

asb, 2019-07-11:
In the LLVM backend we did LOAD, LOAD_FP, MISC_MEM, ...

I don't feel strongly - anything that's logical.

closed 2019-07-11 by towoe