lowRISC/ibex #140: Simulation  fail because of asm_file

haykp, 2019-07-11:
In the sim bash script, it reads asm_test file,
but there is no such file

Can someone help to get that file?

I also tried to use iss_sim to generate the bin file, but still no success((

imphil, 2019-07-11:
Unfortunately the DV part of Ibex is currently rather undocumented. What commands did you run? Maybe @taoliug can help you out then, who wrote most of this code.

imphil, 2019-07-11:
To make sure others don't stumble across the same issues I've opened https://github.com/lowRISC/ibex/issues/141 to get the DV flow documented.

taoliug, 2019-07-11:
I will take care of the DV flow documentation. Meanwhile, can you try just "make" under dv/uvm directory? 
Please make sure you have ${RISCV_TOOLCHAIN} setup properly to point to the directory where RISC-V gcc toolchain and spike live.
https://github.com/lowRISC/ibex/blob/master/dv/uvm/Makefile#L7

taoliug, 2019-08-02:
Please check the DV flow document and let me know if you have any issue
https://github.com/lowRISC/ibex/tree/master/dv/uvm#overview

closed 2019-08-02 by taoliug