lowRISC/ibex #142: Problem with LBU instruction

jeanlouismaquet, 2019-07-11:
Hello, 

The problem appears during the lbu instruction.
With the new code ( **_ibex_load_store_unit.sv_** ) , the address is not correct.

Files provided:
Archive:  **_problem_with_lbu.zip_**
     bootloader_ibex.objdump            dump of the elf file
     cmd_AES_CBC_zero_riscy.log       results generated by the tracer module , for the zero_riscy   
     cmd_AES_CBC_ibex.log                results generated by the tracer module , for the new ibex  
     cmd_AES_CBC_ibex_old.log         results generated by the tracer module , for the old ibex 

### **In the objdump file, we have:**

     e60:       4741                    li      a4,16
     e62:       d43e                    sw      a5,40(sp)
     e64:       07b1                    addi    a5,a5,12
     e66:       993e                    add     s2,s2,a5
     e68:       1001f7b7                lui     a5,0x1001f
     e6c:       84f7c783                lbu     a5,-1969(a5) # 1001e84f <__bss_start+0x7>
     e70:       00b4                    addi    a3,sp,72
     e72:       c83a                    sw      a4,16(sp)
     e74:       ca02                    sw      zero,20(sp)

   a5 (x15) = 0x1001f000
   0x1001f000 - d1969 = **0x1001e84f**
   

### **_In the cmd_AES_CBC_zero_riscy.log and cmd_AES_CBC_ibex_old.log files_** (from the tracer module), we have:

 ```
00000e62 02f12423 sw               x15, 40(x2)         x15:00000078  x2:1001f600  PA:1001f628
 00000e64 00c78793 addi             x15, x15, 12        x15=00000084 x15:00000078
 00000e66 00f90933 add              x18, x18, x15       x18=1001f6c0 x18:1001f63c x15:00000084
 00000e68 1001f7b7 lui              x15, 0x1001f000     x15=1001f000
 00000e6c 84f7c783 lbu              x15, -1969(x15)     x15=00000001 x15:1001f000  **_PA:1001e84f_**
 00000e70 04810693 addi             x13, x2, 72         x13=1001f648  x2:1001f600
 00000e72 00e12823 sw               x14, 16(x2)         x14:00000010  x2:1001f600  PA:1001f610
 00000e74 00012a23 sw               x0, 20(x2)           x2:1001f600  PA:1001f614
```

 x15:1001f000 - 1969 = **_PA:1001e84f_**     =>   OK
 
 
### _In the cmd_AES_CBC_ibex.log, we have :_
 
```
 00000e64 00c78793 addi             x15, x15, 12        x15=000000f4 x15:000000e8
 00000e66 00f90933 add              x18, x18, x15       x18=1001f730 x18:1001f63c x15:000000f4
 00000e68 1001f7b7 lui              x15, 0x1001f000     x15=1001f000
 00000e6c 84f7c783 lbu              x15, -1969(x15)     x15=00000000 x15:1001f000  PA:1001e84c
 00000e70 04810693 addi             x13, x2, 72         x13=1001f648  x2:1001f600
 00000e72 00e12823 sw               x14, 16(x2)         x14:00000010  x2:1001f600  PA:1001f610
 00000e74 00012a23 sw               x0, 20(x2)           x2:1001f600  PA:1001f614
 00000e76 02012023 sw               x0, 32(x2)           x2:1001f600  PA:1001f620
```
 
 
 x15:1001f000 - 1969 = **PA:1001e84c**     =>   Error
 
 
 
###  **_Workaround, in the  ibex_load_store_unit.sv:_**
 
 
  // output data address must be word aligned
  // assign data_addr_w_aligned = {data_addr[31:2], 2'b00};
  **_assign data_addr_w_aligned = {data_addr[31:0]};_**
  // output to data interface
  assign data_addr_o   = data_addr_w_aligned;
  
With this modification, my simulation is successful. 
 
Remark: 
I've a LSU2AHB module that can manage access bytes.
Addresses do not need to be aligned word by word.

Best regards,

jean-Louis
[problem_with_lbu.zip](https://github.com/lowRISC/ibex/files/3381685/problem_with_lbu.zip)

towoe, 2019-07-11:
Hi Jean-Louis, thanks for opening this issue, could you clarify what you mean by old and new ibex?

vogelpi, 2019-07-11:
Thanks for reporting this bug @jeanlouismaquet .

There is obviously a problem in the read data alignment for byte accesses. I will have a look and work out a fix that also works when the memory only supports word-aligned accesses. I will get back to you.

vogelpi, 2019-07-11:
Hi @jeanlouismaquet ,
I had a look and I cannot produce this behavior with the current version of Ibex. Can you please confirm the commit ID with which you experience this behavior?

vogelpi, 2019-07-16:
I could not reproduce the reported behavior with the current version of Ibex. I am thus closing the issue for now.

closed 2019-07-16 by vogelpi