lowRISC/ibex #147: New tracer doesn't work as expected

taoliug, 2019-07-11:
I am running regression with new tracer with RVFI interface, seems to me the generated trace is not as expected.  Looks like the logic used to generate rvfi_valid_int is one cycle off. The decoded instruction is not captured properly.

[image]

Related code:
https://github.com/lowRISC/ibex/blob/master/rtl/ibex_core.sv#L667

towoe, 2019-07-12:
@taoliug thanks for reporting that. Obviously this is not correct, but I am not yet sure what the correct solution is. My current test is not showing this, so I will need to elaborate here. Despite that I guess @vogelpi can help as he has changed the IF-ID interface. I had a look at the signal change but did not catch this. My first assumption would be to change `rvfi_valid_int` to `instr_ret`.

vogelpi, 2019-07-12:
@taoliug thanks for reporting the bug here. I am sorry @towoe , but I won't be able to fix it as I am not able to test the tracer (it cannot be used with Verilator). But maybe I can explain what changes I did and how they possibly influence RVFI.

I am not 100% sure what the `rvfi_valid_int` is used for. If I understand it correctly, it is used to sample new data for RVFI and thus also the tracer. Previously, it was depending on the signal `id_valid` inside the controller and I think how this signal was generated was wrong. It was not really indicating whether the ID stage had valid input or output data. I changed this signal name to `id_out_valid` to indicate when the ID stage has valid output data.

Now for RVFI and the tracer, I think what you want is the signal that indicates when the ID stage has valid input data and you only want this during one cycle to only sample once for multicycle instructions. To achieve this, you can use the signal `instr_new_id`. This signal is indicating when the ID stage gets to see a new instruction and it is the pipeline register write enable signal `if_id_pipe_reg_we` delayed by one cycle.

I also do not think you need to do the AND with `illegal_c_insn_id`, otherwise the the tracer will not record illegal compressed instructions (but illegal non-compressed instructions would be recorded now).

Please let me know if you need more info or if I can somehow assist you.
Pirmin

towoe, 2019-07-12:
@vogelpi thanks a lot for the help!

`rvfi_valid_int` is used to for `rvfi_valid` to signal all output data are valid in this cycle with a retired instruction. You are right to assume it should only be asserted for one cycle in a multicycle instruction.

As I understand it, `instr_new_id` marks the "beginning" and `instr_ret` the "end" of an instruction?
The problem I see with `instr_ret` is that it is only asserted if a new valid instruction is present, so this could be problematic if there is a stall in the instruction input?
As `id_out_valid` depends on `instr_valid_id_o` this would be asserted for more than on cycle in a multicycle instruction?
So `instr_new_id` might be the correct solution here :)

Omitting `illegal_c_insn_id` could be problematic for the RISC-V formal verification, but this can not be tested at the moment. 

@taoliug could you just test the following?
```
assign rvfi_valid_int = id_out_valid;
```

vogelpi, 2019-07-12:
You are welcome @towoe .

> As I understand it, `instr_new_id` marks the "beginning" and `instr_ret` the "end" of an instruction?

Yep.
> The problem I see with `instr_ret` is that it is only asserted if a new valid instruction is present, so this could be problematic if there is a stall in the instruction input?

Yes I see. Maybe the best would to regenerate `instr_ret` inside the ID stage based on what is going on in the ID stage, and not wait for the next instruction to arrive. I will do that and prepare a PR for you.

> As `id_out_valid` depends on `instr_valid_id_o` this would be asserted for more than on cycle in a multicycle instruction?

Yep.

towoe, 2019-07-12:
Thanks @vogelpi for putting in the work for a PR, I think your idea to regenerate `instr_ret` is the right way to go here.
The tracer might work with `instr_new_id`, but I just saw that with a compressed instruction `rvfi_valid` will not be correctly timed.

vogelpi, 2019-07-12:
Okay good. I created the PR #150 and it is waiting to be reviewed.

towoe, 2019-07-12:
I looked at the PR and noticed that I will need to work on RVFI so that it will be correct again.
For the tracer part I think it should be okay to use `instr_new_id` at the moment for `rvfi_valid_int`, until the changes to RVFI are implemented.

taoliug, 2019-07-13:
> @vogelpi thanks a lot for the help!
> 
> `rvfi_valid_int` is used to for `rvfi_valid` to signal all output data are valid in this cycle with a retired instruction. You are right to assume it should only be asserted for one cycle in a multicycle instruction.
> 
> As I understand it, `instr_new_id` marks the "beginning" and `instr_ret` the "end" of an instruction?
> The problem I see with `instr_ret` is that it is only asserted if a new valid instruction is present, so this could be problematic if there is a stall in the instruction input?
> As `id_out_valid` depends on `instr_valid_id_o` this would be asserted for more than on cycle in a multicycle instruction?
> So `instr_new_id` might be the correct solution here :)
> 
> Omitting `illegal_c_insn_id` could be problematic for the RISC-V formal verification, but this can not be tested at the moment.
> 
> @taoliug could you just test the following?
> 
> ```
> assign rvfi_valid_int = instr_new_id;
> ```

I tried this together with PR #150.

Here's the new trace from ibex

              231000              17 80000080 0000a801 INVALID                             
              253000              28 80000090 f4b0a037 lui              x0, 0xf4b0a000     
              255000              29 80000094 b5f00013 addi             x0, x0, -1185      
              263000              33 80000098 5f7d30b7 lui              x1, 0x5f7d3000       x1=0x5f7d3000
              265000              34 8000009c 04608093 addi             x1, x1, 70           x1=0x5f7d3046  x1:0x5f7d3000
              327000              65 800000a0 58e5a137 lui              x2, 0x58e5a000       x2=0x58e5a000
              335000              69 800000a4 43d10113 addi             x2, x2, 1085         x2=0x58e5a43d  x2:0x58e5a000
              357000              80 800000a8 800001b7 lui              x3, 0x80000000       x3=0x80000000
              379000              91 800000ac 00004201 INVALID                             

Trace from spike:

core   0: 0xffffffff80000080 (0x0000a801) j       pc + 16
3 0x80000080 (0xa801)
core   0: 0xffffffff80000090 (0xf4b0a037) lui     zero, 0xf4b0a
3 0x80000090 (0xf4b0a037)
core   0: 0xffffffff80000094 (0xb5f00013) li      zero, -1185
3 0x80000094 (0xb5f00013)
core   0: 0xffffffff80000098 (0x5f7d30b7) lui     ra, 0x5f7d3
3 0x80000098 (0x5f7d30b7) x 1 0x5f7d3000
core   0: 0xffffffff8000009c (0x04608093) addi    ra, ra, 70
3 0x8000009c (0x04608093) x 1 0x5f7d3046
core   0: 0xffffffff800000a0 (0x58e5a137) lui     sp, 0x58e5a
3 0x800000a0 (0x58e5a137) x 2 0x58e5a000
core   0: 0xffffffff800000a4 (0x43d10113) addi    sp, sp, 1085
3 0x800000a4 (0x43d10113) x 2 0x58e5a43d
core   0: 0xffffffff800000a8 (0x800001b7) lui     gp, 0x80000
3 0x800000a8 (0x800001b7) x 3 0x80000000
core   0: 0xffffffff800000ac (0x00004201) li      tp, 0

You can see the binary part actually matches, but ibex tracer decodes the instruction as invalid. The problem seems to be related to compressed instructions

towoe, 2019-07-15:
Thanks @taoliug for testing this again.
Just for reference, using `assign rvfi_valid_int = instr_new_id;` does not require PR https://github.com/lowRISC/ibex/pull/150.

Regarding the invalid, this is indeed cause by compressed instructions.
RVFI requires the output of the 'original' compressed instruction, but the tracer is not able to decode compressed instructions.

vogelpi, 2019-07-15:
Thanks for your feedback @taoliug and @towoe , I think we should add support for compressed instructions in the tracer by feeding also the uncompressed instruction through RVFI. This part is tracked in #154 .

As for #150 , even though it is not strictly needed as pointed out by @towoe , we should merge it. In my opinion this is the only proper way to generate this signal.

towoe, 2019-07-15:
> As for #150 , even though it is not strictly needed as pointed out by @towoe , we should merge it. In my opinion this is the only proper way to generate this signal.

I am currently working on https://github.com/lowRISC/ibex/issues/151 which makes use of your PR. So yes, I agree that we should merge it.

vogelpi, 2019-07-15:
Thanks for your feedback @towoe , that's good to know.

commit d059979a (2019-07-16): RVFI based on core state (#151, fix #147)

The RVFI implementation make use of signals from the decoder and
controller to detect the state of the processor.
Especially the signal for a new and retired instruction.
closed 2019-07-16 by towoe