lowRISC/ibex #153: dpc is not updated properly if debug_req_i is a one cycle pulse

taoliug, 2019-07-14:
As you can see from below waveform, debug_req_i is modeled as a pulse interrupt which is only asserted for one cycle. ibex can enter debug mode and start fetching instructions from DmHaltAddr. However, dpc is not updated properly (keep at default value 0). I cannot find in the RISC-V debug spec that whether the halt request should be a level or pulse signal, can you help to clarify?  

[image]

vogelpi, 2019-07-15:
Hi @taoliug ,
thanks for opening this issue. I do not think that having `debug_req_i` asserted for just one clock cycle is realistic. However, I still think there is a bug that needs to be fixed.

The same behavior will occur whenever the `debug_req_i` is deasserted during the cycle the controller wants to enter debug mode. Since it can take a very long time for the core to actually enter debug mode upon receiving a request (the spec just mandates the core to enter debug mode withing 1 second), it can very well be that the `debug_req_i` is deasserted if the core does not enter debug mode for some time.

There are basically two options to fix this:
1. Once the controller decides to halt and therefore jumps into `DBG_TAKEN_ID` state, it enters debug mode even if `debug_req_i` is deasserted one cycle later. That is we save the `debug_req_i` signal when jumping into `DBG_TAKEN_ID`.
2. The controller keeps checking the `debug_req_i` signal when in `DBG_TAKEN_ID` stage. If it is deasserted, it re-enters `DECODE` state without entering debug mode.

What do you think is the better solution @imphil? Or do you have any other ideas?
Once we have made a decision, I will prepare a fix and create a PR.

imphil, 2019-07-15:
On the expected behavior of debug_req_i:

`debug_req_i` is a level signal which corresponds to the `haltreq` register in the debug module. This register is set and reset by explicit writes from the debugger software to the debug module. The core acknowledges receiving the `debug_req_i` signal by entering debug mode and 

The actual process is described in section "B.3 Halting" of the [debug spec](https://github.com/riscv/riscv-debug-spec/blob/master/riscv-debug-draft.pdf).

1. Debugger sets haltreq register in the debug module (DM). This sets debug_req_i.
2. The core gets debug_req_i and jumps into the memory region of the debug rom
3. The software code in the debug rom resets the halted register. (https://github.com/pulp-platform/riscv-dbg/blob/master/debug_rom/debug_rom.S#L50, https://github.com/pulp-platform/riscv-dbg/blob/master/src/dm_mem.sv#L208)
4. The debugger polls the halted register and waits until it is set.
5. The debugger can now reset haltreq, which in turn deasserts debug_req_i

So the whole process is rather long and involves a couple host - chip round trips.

On the bug itself:
Even if a one-cycle debug_req_i isn't realistic, it still should behave sane, i.e. it is most likely ignored. Option 1 or 2 sound reasonable, I'd go for option 2.

vogelpi, 2019-07-15:
Okay, thanks for outlining the procedure and your feedback @imphil . I will implement Option 2 then.

PR #157 (2019-07-15): Fix dpc CSR not updated
This PR lets the core also check `debug_req_i` and `debug_single_step_i` while entering debug mode to avoid entering debug mode without setting up the debug CSRs properly (which anyway depend on these signals), if these signals go low during the procedure.

This resolves #153 .
commit a775abd9 (2019-07-15): Fix dpc CSR not updated (#157)

* Controller: remove impossible condition for `DBG_TAKEN_IF`

There is no way to jump into `DBG_TAKEN_IF` because of an EBREAK
instruction. Thus, this case also does not need to be checked.

* Controller: do not enter debug when `debug_req_i` goes low

With this commit, the core is prevented from entering debug mode when
the debug request signal is deasserted during that procedure.
Previously, the core would still enter debug mode but not updating the
debug CSR.

This resolves #153 reported by @taoliug.

* Update ibex_controller.sv
closed 2019-07-15 by taoliug