lowRISC/ibex #154: Tracer support for compressed instructions

towoe, 2019-07-15:
The tracer reuses the RVFI signals. `rvfi_insn` holds the current instruction, but for the RISC-V formal verification this needs to be the instruction which is inserted into the core.
For compressed instructions the core will decode those to uncompressed instruction, which are then used internally.

The tracer currently prints `INVALID` for compressed instructions (https://github.com/lowRISC/ibex/issues/147#issuecomment-511083710) as it has no support for compressed instructions. 

For the tracer we can either add an additional output signal with the uncompressed instruction or add support in the tracer for decoding compressed instructions.
@imphil do you have an opinion here?

vogelpi, 2019-07-15:
Since we currently have no possibility to actually test the tracer, I suggest to add the additional output signal with the uncompressed instruction. But let's wait on @imphil's opinion.

towoe, 2019-07-15:
Thanks for the reply @vogelpi.
I also tend to just use an additional signal as this is also the signal which the core does rely on, so it makes sense to use it for the tracer.

vogelpi, 2019-07-15:
I just discussed with @imphil . We should extend the RVFI with the uncompressed instruction. This is a custom extension to RVFI, but no one is forced to use it. It should be fine.

towoe, 2019-07-15:
Ok thanks!

closed 2019-07-16 by towoe