lowRISC/ibex #162: Load instructions that cause an error still write to registers

ivanmgribeiro, 2019-07-16:
Hi,
There seems to be a bug in Ibex when performing a read from memory which returns an error.

I would expect that when the memory slave sets data_err high, the core would not write the data that was read in into a register, but it appears to still be written to the register regardless.

What i observe is that the data is still written into the register.

I'm using the following instructions:
```
addi r2, r0, 1
lw r2, r2[2]
addi r3, r2, 1
```
When using the load word instruction, the register is overwritten with 0x0, but when using load halfword or load byte, it seems to actually read what is returned by memory and store it in the register.

My memory slave is configured such that accesses outside of the range 0x80000000 - 0x80007fff will set data_err high.

I've performed a fix that seemed to work locally by changing the line
https://github.com/lowRISC/ibex/blob/b7ef64b62da74afe52bb40165e17a7e87539d47a/rtl/ibex_id_stage.sv#L283
to
```
assign regfile_we = (illegal_csr_insn_i || !instr_executing || lsu_load_err_i) ? 1'b0          :
```

This fix seems to work with other failing loads as well, but I'm not sure that it conforms to your style guide or whether it should be done here or at some other point in the code.

vogelpi, 2019-07-16:
Hi @ivanmgribeiro ,
thanks for reporting this bug. I will prepare a fix for and create a PR.
Best regards,
Pirmin

imphil, 2019-07-17:
General note: @ivanmgribeiro you are invited to create Pull Requests with the fix directly if you already have a patch! Let me know if you run into issues while doing that.

vogelpi, 2019-07-17:
Hi @ivanmgribeiro ,
regarding the differing behavior on what is actually written to the register file depending on the type of load operation: This happens as depending on type and address offset, the LSU issues two load operations and combines the result of the two loads into one value for the register file.

For aligned word loads, byte loads and half-word loads with address offset != `2'b11`, the LSU just performs one load and returns the value obtained from memory. For misaligned loads, the LSU combines the value obtained from memory with an internal register, which is reset to zero.

Best regards,
Pirmin

PR #169 (2019-07-18): ID/EX stage: do not write to register file upon load errors
This commit fixes the write back FSM to not store values returned from
memory to the register file when the LSU is reporting a load error.

This bug was reported by @ivanmgribeiro. This commit resolves #162.
commit 63702b3e (2019-07-18): ID/EX stage: do not write to register file upon load errors

This commit fixes the write back FSM to not store values returned from
memory to the register file when the LSU is reporting a load error.

This bug was reported by @ivanmgribeiro. This commit resolves #162.
closed 2019-07-18 by vogelpi