lowRISC/ibex #163: 7d10b363 is not decoded as illegal instruction

taoliug, 2019-07-16:
Based on the RISC-V instruction encoding, 7d10b363 doesn't map to any legal instruction. The trace log shows the instruction as INVALID. However, the id_stage doesn't assert the illegal_insn signal, and the core continues executing the instructions without jump to exception handling routine. Looks like the core can only detect illegal compressed instructions and illegal CSR instructions.

Spike log

```
core   0: 0xffffffff8000faaa (0x7d10b363) unknown
core   0: exception trap_illegal_instruction, epc 0xffffffff8000faaa
core   0:           tval 0x0000000000000000
core   0: 0x000000008001d000 (0x00007119) addi    sp, sp, -128
```

ibex log
```
            22039000           10921 8000faa8 40b6d693 srai             x13, x13, 0x40b     x13=0x00000000 x13:0x00000000
            22041000           10922 8000faaa 7d10b363 INVALID                             
            22047000           10925 8000faae 001d0d13 addi             x26, x26, 1         x26=0x5fab433a x26:0x5fab4339

```

taoliug, 2019-07-16:
This could be a general issue that how the core handles illegal non-compressed instructions. There are certain encoding don't map to any legal instruction. I am seeing a few other failure signatures in the same category.

vogelpi, 2019-07-16:
Hi @taoliug ,
thanks for reporting this bug. Looking at the source code I guess only branch instructions (OPCODE = `7'b63`) are affected by this behavior. Is that right?
I think the problem is that for branch instructions, the illegal instruction output signal is accidentally only driven during the first cycle while the branch can take a second cycle. Since during the second cycle (and when the controller is actually allowed to jump to the exception handler) the illegal instruction signal is no longer set. The core just continues and the exception is undetected. I will prepare a fix for this tomorrow.

Are there any other instructions encodings that you see showing this behavior for?

taoliug, 2019-07-16:
Here's another example 1613ac33, it's close to SLTU, but instr[31:25] !=0

vogelpi, 2019-07-17:
> Here's another example 1613ac33, it's close to SLTU, but instr[31:25] !=0

That's another bug in the decoder. It does not cause illegal instruction exceptions for register-register ALU operations as long as `instr[28]` is set. I will fix this as well.

closed 2019-07-17 by vogelpi