lowRISC/ibex #164: Debug request causes the same instruction to be executed twice

taoliug, 2019-07-16:
In this test, debug_req is asserted while the core is executing instruction at PC 0x8000056a. dpc is saved to the current PC value rather than the next instruction. As a result, the instruction at 0x8000056a is executed again after exiting from debug mode(dret). 

ibex log
```
             8487000            4145 8000056a 00cf9f93 slli             x31, x31, 0xc       x31=0x55508000 x31:0x9d655508
             8595000            4199 80000040 3780e06f jal              x0, 58232          
             8639000            4221 8000e3b8 7b200073 dret                                
             8683000            4243 8000056a 00cf9f93 slli             x31, x31, 0xc       x31=0x08000000 x31:0x55508000
```
[image]

taoliug, 2019-07-16:
Looks like dpc value is not shown properly on the waveform, it is 0x8000056a after entering debug mode.

vogelpi, 2019-07-17:
Thanks @taoliug for reporting this bug. You are right, it should be the address of the next instruction that is saved when entering debug mode upon a halt request. I will fix this.

PR #167 (2019-07-17): Controller: save next PC to dpc CSR upon a halt request
This commit changes the controller to save the PC of the next
instruction to the dpc CSR when entering debug mode upon a halt
request instead of the PC of the current instruction (only valid
for EBREAK instructions).

This bug was reported by @taoliug. This resolves #164.
commit 010985a3 (2019-07-17): Controller: save next PC to dpc CSR upon a halt request

This commit changes the controller to save the PC of the next
instruction to the dpc CSR when entering debug mode upon a halt
request instead of the PC of the current instruction (only valid
for EBREAK instructions).

This bug was reported by @taoliug. This resolves #164.
closed 2019-07-17 by vogelpi