lowRISC/ibex #168: Check behavior of exceptions and special instructions related to debug mode

vogelpi, 2019-07-17:
Currently, exceptions and special instructions are handled independent of whether the core is in debug mode or not. This is not correct for a couple of cases. For example, the Debug Spec v0.13.2 states on p.39 that while in debug mode:
- WFI acts as a NOP
- Exceptions do not update any registers such as `mcause`, `mepc`, `mtval`, `dpc` and `mstatus`

Similarly, the DRET instruction must lead to an illegal instruction exception when the core is not in debug mode (p.41).

imphil, 2019-07-17:
> WFI acts as a NOP. Currently the core just goes to sleep anyway.

Really? https://github.com/lowRISC/ibex/blob/010985a39555bbbee969296e1da4f3bd0b502e63/rtl/ibex_controller.sv#L251-L254

vogelpi, 2019-07-17:
Oh, yes you are right. I only saw that in the `FLUSH` state, the core jumps into `SLEEP` when seeing a WFI.

imphil, 2019-07-17:
This behavior is tested by the RISC-V Debug compilance test in openocd, so I'm pretty confident that works.

vogelpi, 2019-08-29:
We have now a separate issue lowrisc/ibex#270 for DRET causing an illegal instruction exception if the core is not in debug mode (p.41).

vogelpi, 2019-08-29:
The DRET behavior has been solved.

PR #475 (2019-11-19): Fix behavior of exceptions in debug mode, document debug CSRs
This PR consists of three small commits which implement the following changes:
- Exceptions in debug mode shall not modify any CSRs. This resolves #168 and is related to #365.
- Document debug CSRs. This resolves #307.
- Update documentation of `mhartid` CSR. Previously, the doc was still describing the cluster and core ID bit fields.
closed 2019-11-19 by vogelpi