lowRISC/ibex #175: Illegal load instruction causes data written to GPR

taoliug, 2019-07-18:
ibex log
```
           551503000          275653 8000f926 0020ff83 INVALID                              x31=0x0000002d PA:0x0001b8ee  load:0x0000002d
           551505000          275654 8000f92a 0c8cb4ab INVALID                             
           551513000          275658 80000000 0001d06f jal              x0, 118784         
```
0020ff83 is an illegal load instruction (bit[14:12] = 3'b111), ibex detects it correctly. However, the read data is still written to the register.

[image]

taoliug, 2019-07-18:
Actually the core decoded it as reg-reg load instruction, I see there's a bug (#25 ) open to remove it as it's not part of the standard.

https://github.com/lowRISC/ibex/blob/master/rtl/ibex_decoder.sv#L335

imphil, 2019-07-19:
Duplicate of #25.

closed 2019-07-19 by imphil