lowRISC/ibex #181: Compile issue while running ibex_core_tracer example

raulbehl, 2019-07-23:
Hi, 

I am running into compile issues while trying to run the `ibex_core_tracer` example using `fusesoc`.  I guess the issue is in the way `timescale` value is specified on the command-line. 

Updating the timescale option from `timescale=1ns/1ns` to `timescale 1ns/1ns` should fix it.

towoe, 2019-07-24:
Hi, thanks for reporting this. What tool and which version are you using? I did a quick check and I guess at some point both options have been a valid format.

raulbehl, 2019-07-24:
I tried this out with Modelsim:

`Model Technology ModelSim ALTERA vlog 10.1e Compiler 2013.06 Jun 12 2013`

raulbehl, 2019-07-25:
I guess Questasim supports the "=" in between the timescale directive and the value.

towoe, 2019-07-25:
> I guess Questasim supports the "=" in between the timescale directive and the value.

Unfortunately I can not test this.

And regarding your Modelsim version, in general I would not change something to support an old version of a tool, but if this is something that works then for all versions I think it makes sense.
The documentation on this is rather thin, so I guess it is based more on tests.
And for reference, the original option was adopted from a [fusesoc tutorial](https://fusesoc.readthedocs.io/en/latest/tutorials/3-tool_options.html).

raulbehl, 2019-07-25:
Can you point me to the recommended version? I have tried it on `Model Technology ModelSim SE vlog 10.3a Compiler 2014.04 Apr 13 2014` which also doesn't support it.

towoe, 2019-07-25:
The general recommendation would be to use the latest version.
But we can also use the different format here if it does not break with newer releases.
If you want you can create a pull request for this?

imphil, 2019-07-29:
Fixed in https://github.com/lowRISC/ibex/pull/192

closed 2019-07-29 by imphil