lowRISC/ibex #182: Does an RV32I (without C) misaligned fetches

eroom1966, 2019-07-23:
Hi
Reading the document section on Instruction Fetch, I understand the following

> Misaligned instruction fetches are handled by performing two separate word-aligned instruction fetches. Internally, the core can deal with both word- and half-word-aligned instruction addresses to support compressed instructions. The LSB of the instruction address is ignored internally.

Is this also true for an RV32I which does NOT support Compressed instructions ?
ie, I could jump to an address on a 2-byte boundary without taking an exception ?

Thx
Lee

vogelpi, 2019-07-23:
Hi Lee,

> Is this also true for an RV32I which does NOT support Compressed instructions ?
> ie, I could jump to an address on a 2-byte boundary without taking an exception ?

No, for RV32I configurations that do NOT support the C extension, instructions must be aligned to 32 bits. The RISC-V Spec v2.2 points this out in Section 16.1, p.95: 

"The C extension allows 16-bit instructions to be freely intermixed with 32-bit instructions, with the latter now able to start on any 16-bit boundary, i.e., IALIGN=16. With the addition of the C extension, no
instructions can raise instruction-address-misaligned exceptions. Removing the 32-bit alignment constraint on the original 32-bit instructions allows significantly greater code density."

So I think your understanding is correct and this is in line with the Priv Spec.

Regarding Ibex: It supports RV32IMC, RV32EMC, RV32IC, RV32EC - The C extension is always enabled. So everything should be fine.

Best regards,
Pirmin

eroom1966, 2019-07-23:
> Regarding Ibex: It supports RV32IMC, RV32EMC, RV32IC, RV32EC - The C extension is always enabled. So everything should be fine.

of course, thanks

closed 2019-07-23 by eroom1966