lowRISC/ibex #183: Exceptions : load/store access fault

eroom1966, 2019-07-23:
In the documentation:-
ibex can trigger an exception due to the following exception causes:
- Load access fault
- Store access fault

Could you give an example of how either of these exceptions could be raised ?

My understanding is that these exceptions would be generated either from a Virtual Memory Fault, or a PMP Fault, neither of these units are present in ibex

Thx
Lee

vogelpi, 2019-07-23:
Hi Lee,

you are right. Ibex currently does not support Virtual Memory or PMP. However, load/store access faults can also be generated by the downstream interconnect or the device being accessed. For example, if the core tries to access an address region that the bus cannot resolve or if the core tries to write to a register inside a peripheral that is read only.

Best regards,
Pirmin

eroom1966, 2019-07-23:
Hi Pirmin
Thanks, and just for final clarification, I think you are referring to the state of the valid/err signals on the interface ?
Thx
Lee

vogelpi, 2019-07-23:
Yes, in Ibex such a condition is signaled with `data_err_i` being high together with `data_rvalid_i`.

closed 2019-07-24 by vogelpi