lowRISC/ibex #187: Tracer compile issue: irq_ack_o

taoliug, 2019-07-24:
Looks like this port is removed recently, but still connected to the tracer.

/usr/local/google/home/taliu/ibex_dbg/ibex/rtl/ibex_core.sv, 645
  Identifier 'irq_ack_o' has not been declared yet. If this error is not 
  expected, please check if you have set `default_nettype to none.

https://github.com/lowRISC/ibex/blob/master/rtl/ibex_core.sv#L645

vogelpi, 2019-07-24:
Hi @taoliug ,
thanks for reporting this issue. Also the port list in `ibex_core_tracing.sv` needs to be fixed.

I will prepare a fix for that.
Best regards,
Pirmin

vogelpi, 2019-07-24:
I am also adding @towoe as he might have some input on this.

@taoliug , I removed the port from the tracer and switched to the new interrupt interface in several remaining files. You can have a look here: https://github.com/vogelpi/ibex/tree/fix-interrupt-interface

To me, it seems that the `irq_ack_o` signal was only used in RVFI and the tracer for UVM purposes. But I am not sure if you actually had a test running for this. Nevertheless, I did not dare to touch the UVM files. It seems that there is some infrastructure for testing this interrupt interface. The following files are affected:

- `dv/uvm/common/irq_agent/irq_if.sv`
- `dv/uvm/common/irq_agent/irq_master_driver.sv`
- `dv/uvm/common/irq_agent/irq_monitor.sv`
- `dv/uvm/tb/core_ibex_tb_top.sv`

Could you have a look at these and replace/adapt accordingly? In general, the interrupt interface of the core is now much simpler. The core has only level-sensitive interrupt inputs and there is no acknowledge signal.

Best regards,
Pirmin

taoliug, 2019-07-24:
Yes, I have a PR locally to fix it. I think it's OK to push the design changes in, and I will submit the DV changes short after.

vogelpi, 2019-07-24:
Okay thanks for your feedback @taoliug . This sounds like a plan.

vogelpi, 2019-07-24:
Alright, PR #188 is merged. Please let me know when you want to me to review your PR or once it is merged.

taoliug, 2019-07-24:
TB update is checked in, closing this bug.

closed 2019-07-24 by taoliug