lowRISC/ibex #197: Ibex tracer - compressed instructions

udinator, 2019-07-30:
Hi,
Would it be possible to have the Ibex tracer indicate that an instruction is compressed, and if so, log the compressed instruction name? This would bring the log file closer in similarity to Spike log files, and would make debugging regression failures a bit easier to do.
Not sure if this complies with RVFI spec though.
Thanks!
Udi

imphil, 2019-07-31:
Sure, that's possible. We have the `rvfi_insn` signal, which contains the original (compressed) instruction. (The non-standard `rvfi_insn_uncompressed` signal always contains the uncompressed instruction. This one is currently used by ibex_tracer.)

We have two options going forward:
- Either you open a PR which extends `ibex_tracer` to include the compressed instruction.
- Or you let me know how you'd like the logs to look and I'll implement it.

udinator, 2019-07-31:
Current ibex trace logs are below:
[image]
[image]

I was just referring to denoting the instructions as compressed (such as c.addi, etc...) like Spike logs do:
[image]
[image]

I was just thinking it could be efficient for the ibex logs to show the compressed instruction name (c.li, c.addi, c.lui, etc...) to match what Spike shows and give a better idea of which instructions are compressed or not. Thanks!

commit 60de915d (2019-09-06): Adding Compressed Instruction support in tracer

Added compressed instruction decoder in the tracer to correctly
trace compressed instructions with their mnemonics. Fixes #197
closed 2019-09-06 by imphil