lowRISC/ibex #198: No trap on misaligned jump

ivanmgribeiro, 2019-07-30:
Ibex doesn't seem to be trapping when a jump or branch instruction tries to jump to a location that isn't 2-byte aligned (LSB is 1).

ivanmgribeiro, 2019-07-31:
Sorry, I misunderstood the spec, it appears that implementations that support compressed instructions don't throw Instruction-address-misaligned.

imphil, 2019-07-31:
You're trying to jump to a instruction which is only byte-aligned (LSB = 1), so it's illegal for both compressed (16b aligned) and non-compressed (32b aligned) instructions.

Let me re-read the spec here, but it would feel natural to trap ...

imphil, 2019-07-31:
I had a look at the spec, and Ibex is working just fine. 

- Jumps are always given as multiples of 2 bytes, so you can only jump to 16b-aligned addresses. (Your initial question suggests you're jumping to a byte-aligned address. How did you do that?)
- With compressed instructions supported (which we unconditionally do in ibex), any jump address is a valid address.

closed 2019-07-31 by imphil

ivanmgribeiro, 2019-07-31:
>(Your initial question suggests you're jumping to a byte-aligned address. How did you do that?)

From what I understand of the spec, the JALR instruction allows you to pass in any immediate, not necessarily a 2-byte aligned one (unlike the JAL and BRANCH instructions). However the JALR then adds this to the register contents to get the target and sets the LSB of the jump target to 0. This means I could try to jump to a location with LSB=1 using the instruction but it should then set the LSB to 0 which it does indeed do.

vogelpi, 2019-08-05:
Just for reference, this topic was discussed some time ago in our Zulip chat here:
https://lowrisc.zulipchat.com/#narrow/stream/198227-ibex/topic/jalr.20clearing.20lsb