lowRISC/ibex #199: interrupt detection

udinator, 2019-07-30:
From the below waveform, our interrupt testbench model pulses interrupts for just one cycle. When ibex is in the SLEEP state and a nonmaskable interrupt is pulsed for one cycle (irq_nm_i), ibex immediately wakes up and jumps to FIRST_FETCH state and starts executing code as normal, but doesn't do any handling of the interrupt, even though it is non-maskable [RISC-V Privileged Spec, section 3.3.4]. Note that handle_irq is is only high for the duration of this interrupt, for one cycle.

[image]

However, in the second waveform below, we have a software interrupt pulsed for one cycle directly after a non maskable interrupt, which is also pulsed just for one cycle. Because these interrupts are back to back, the handle_irq wire is high for two cycles, causing ibex to move to IRQ_TAKEN state, from which it jumps to interrupt handler code elsewhere in the program. However, since the irq_software signal was high while in the IRQ_TAKEN state, ibex only registers the software interrupt by setting exc_cause_o to EXC_CAUSE_IRQ_SOFTWARE_M, and thus appears to ignore that there was a non-maskable interrupt just one cycle before.

[image]

As a result, it would seem that we have to hold interrupts high for two or more cycles for ibex to properly detect and handle any interrupts. However ibex will wake up out of SLEEP state after detecting irq_nm_i pulsed for only one cycle, and will not handle this interrupt. Can you help clarify what is considered correct interrupt detection behavior in these situations?

vogelpi, 2019-08-05:
Hi @udinator ,
thanks for reporting this issue. The main question here is what to do if interrupts are being high for exactly one cycle only.

In short, I think the current implementation of Ibex is fine as we state in the doc that all interrupt lines are level-sensitive and that it is assumed that the interrupt handler signals completion of the handling routine to the interrupt source, e.g., through a memory-mapped register, which then deasserts the corresponding interrupt line. The RISC-V spec does not comment on how long an interrupt signal needs to be set for the interrupt to be taken.

For the long answer, it is best to look at the different interrupt types separately:

1. For normal interrupts (fast, timer, external, software) the RISC-V privileged spec is a bit ambiguous. It says "An interrupt i will be taken if bit i is set in both mip and mie, and if interrupts are globally enabled" [RISC-V Priv Spec v. 1.11, Section 3.1.9, p.31] and further "Only the bits corresponding to lower-privilege software interrupts (USIP, SSIP), timer interrupts (UTIP, STIP), and external interrupts (UEIP, SEIP) in mip are writable through this CSR address; the remaining bits are read-only" [RISC-V Priv Spec v. 1.11, Section 3.1.9, p.29], and that the mip bits for the timer, external, and software interrupts are being deasserted by the timer, PLIC etc. To me, this sounds that it is assumed that these interrupts are all level-sensitive and that they will always be set for more than one cycle.
Thus I think for normal interrupts, the implementation is correct.

2. For the non-maskable interrupt, the spec just says "NMIs cause an immediate jump to an implementation-defined NMI vector running in M-mode regardless of the state of a hart’s interrupt enable bits" [RISC-V Priv Spec v. 1.11, Section 3.4, p.42]. In this context, immediate just means before finishing a currently running interrupt or exception handler. But it does not mean in the very same clock cycle. It is not commented on how long the interrupt is high or the like.
I think that our implementation is okay also here as we state that the interrupt line is level-sensitive and de-asserted externally.

Regarding the WFI instruction, the spec says that software needs to be able to deal with the case if an interrupt wakes up the core but is not taken afterwards: "If the event that causes the hart to resume execution does not cause an interrupt to be taken, execution will resume at pc + 4, and software must determine what action to take, including looping back to repeat the WFI if there was no actionable event" [RISC-V Priv Spec v. 1.11, Section 3.2.3, p.42]. In my view, it is not relevant here whether the interrupt is not taken because it is deasserted after one clock cycle or if interrupts are globally disabled (The operation of WFI must be unaffected by the global interrupt enable).

We could add a register to `irq_nm_i` and `irq_pending_i` to make sure the core also takes interrupts that are high for just one cycle but I think this is not required and would add more complexity (as this registers will need to work on the un-gated clock to also buffer interrupts during a WFI). I would also be interested in the opinion of @imphil regarding this. What do you think?

Best regards,
Pirmin

udinator, 2019-08-05:
Hi @vogelpi,
Thanks for your input. I agree with your thoughts on this matter, it makes sense to me as well that it is assumed that these interrupts should be treated as being level sensitive, especially since a PLIC is ultimately responsible for overall interrupt control.
I don't think that adding registers to latch the two irq_ signals is necessary, and I will modify the interrupt stimulus driver in my testbench to more accurately represent level sensitive interrupts.
Also tagging @taoliug just to include in the discussion.
Best,
Udi

vogelpi, 2019-10-31:
The original question has been answered and the documentation contains the relevant information: all interrupts are level-sensitive and it is assumed that the interrupt handler signals completion of the handling routine to the interrupt source, e.g., through a memory-mapped register, which then deasserts the corresponding interrupt line. The RISC-V spec does not comment on how long an interrupt signal needs to be set for the interrupt to be taken.

I thus closing the issue now.

closed 2019-10-31 by vogelpi